nobody43/zabbix-mini-IPMI #2: Command outputs needed

nobodysu, 2016-12-29:
Please post your OHMR and lmsensors output, from one or more processor boards. Need it for new script version.

nobodysu, 2017-10-03:
OHMR reports could be found here:
https://github.com/openhardwaremonitor/openhardwaremonitor/issues?utf8=%E2%9C%93&q=is%3Aissue%20report

nobodysu, 2017-11-04:
Got lmsensors.
Badly need outputs of `sysctl dev.cpu | grep temperature` on FreeBSD system (after `kldload coretemp` or `kldload amdtemp` of course).

gecon, 2019-05-28:
```
# sensors
k10temp-pci-00c3
Adapter: PCI adapter
Tdie:         +44.2°C  (high = +70.0°C)
Tctl:         +71.2°C

k10temp-pci-00cb
Adapter: PCI adapter
Tdie:         +44.0°C  (high = +70.0°C)
Tctl:         +71.0°C
```

on:
````
# lscpu
Architecture:          x86_64
CPU op-mode(s):        32-bit, 64-bit
Byte Order:            Little Endian
CPU(s):                32
On-line CPU(s) list:   0-31
Thread(s) per core:    2
Core(s) per socket:    16
Socket(s):             1
NUMA node(s):          1
Vendor ID:             AuthenticAMD
CPU family:            23
Model:                 8
Model name:            AMD Ryzen Threadripper 2950X 16-Core Processor
Stepping:              2
CPU MHz:               2657.291
CPU max MHz:           3500.0000
CPU min MHz:           2200.0000
BogoMIPS:              6999.15
Virtualization:        AMD-V
L1d cache:             32K
L1i cache:             64K
L2 cache:              512K
L3 cache:              8192K
NUMA node0 CPU(s):     0-31
Flags:                 fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_good nopl nonstop_tsc cpuid extd_apicid amd_dcm aperfmperf pni pclmulqdq monitor ssse3 fma cx16 sse4_1 sse4_2 movbe popcnt aes xsave avx f16c rdrand lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw skinit wdt tce topoext perfctr_core perfctr_nb bpext perfctr_llc mwaitx cpb hw_pstate sme ssbd sev ibpb vmmcall fsgsbase bmi1 avx2 smep bmi2 rdseed adx smap clflushopt sha_ni xsaveopt xsavec xgetbv1 xsaves clzero irperf xsaveerptr arat npt lbrv svm_
```

nobodysu, 2019-05-28:
At last! Strange output, wonder if it works. I'll test it out, thanks!

nobodysu, 2019-11-25:
@gecon 
Could you please provide `sensors -u` output? Is it one processor or two?

gecon, 2019-11-25:
Here you go

```
# sensors -u
k10temp-pci-00c3
Adapter: PCI adapter
Tdie:
  temp1_input: 48.000
  temp1_max: 70.000
Tctl:
  temp2_input: 75.000

k10temp-pci-00cb
Adapter: PCI adapter
Tdie:
  temp1_input: 47.500
  temp1_max: 70.000
Tctl:
  temp2_input: 74.500
```

I have sent you lscpu in message https://github.com/nobodysu/zabbix-mini-IPMI/issues/2#issuecomment-496481486

It is a single CPU system (not Dual), actually it is one AMD Ryzen Threadripper 2950X 16 Core (so it shows 32 threads in lscpu output).

nobodysu, 2019-11-27:
@gecon 
Would be glad if you test the solution out:
https://github.com/nobodysu/zabbix-mini-IPMI/blob/cpu_refactoring/Linux/mini_ipmi_lmsensors.py

gecon, 2019-11-27:
@nobodysu Hi!

Here is getverb results (using zabbix_get):

``
"THEHOST" mini.cpu.info[cpu0,ID] "k10temp-pci-00c3"
"THEHOST" mini.cpu.info[cpu0,TjMax] "70"
"THEHOST" mini.cpu.temp[cpu0,core1] "43"
"THEHOST" mini.cpu.temp[cpu0,MAX] "43"
"THEHOST" mini.cpu.info[cpu1,ID] "k10temp-pci-00cb"
"THEHOST" mini.cpu.info[cpu1,TjMax] "70"
"THEHOST" mini.cpu.temp[cpu1,core1] "42"
"THEHOST" mini.cpu.temp[cpu1,MAX] "42"
"THEHOST" mini.cpu.temp[MAX] "43"
"THEHOST" mini.cpu.info[ConfigStatus] "NOGPUS"

``

and sample sensors run:

``
# sensors -u
coretemp-isa-0000
Adapter: ISA adapter
Physical id 0:
  temp1_input: 53.000
  temp1_max: 80.000
  temp1_crit: 100.000
  temp1_crit_alarm: 0.000
Core 0:
  temp2_input: 50.000
  temp2_max: 80.000
  temp2_crit: 100.000
  temp2_crit_alarm: 0.000
Core 1:
  temp3_input: 53.000
  temp3_max: 80.000
  temp3_crit: 100.000
  temp3_crit_alarm: 0.000
Core 2:
  temp4_input: 52.000
  temp4_max: 80.000
  temp4_crit: 100.000
  temp4_crit_alarm: 0.000
Core 3:
  temp5_input: 52.000
  temp5_max: 80.000
  temp5_crit: 100.000
  temp5_crit_alarm: 0.000
``

gecon, 2019-11-27:
@nobodysu Seems to be working perfect! 👍

nobodysu, 2019-11-27:
@gecon 
That's great, thanks!

xud6, 2021-03-15:
OHMR report on Ryzen 4650U laptop (Lenovo thinkpad x13 gen1)
```
Open Hardware Monitor Report

--------------------------------------------------------------------------------

Version: 0.9.6.0

--------------------------------------------------------------------------------

Common Language Runtime: 4.0.30319.42000
Operating System: Microsoft Windows NT 6.2.9200.0
Process Type: 32-Bit

--------------------------------------------------------------------------------

Sensors

|
+- Lenovo 20UF0008CD (/mainboard)
|
+- AMD Ryzen 5 PRO 4650U (/amdcpu/0)
|  +- Bus Speed      :  99.8159  99.8159  99.8159 (/amdcpu/0/clock/0)
|  +- CPU Core #1    :  3992.64  3992.64  3992.64 (/amdcpu/0/clock/1)
|  +- CPU Core #2    :  1774.51  1774.51  1774.51 (/amdcpu/0/clock/2)
|  +- CPU Core #3    :  1774.51  1774.51  1774.51 (/amdcpu/0/clock/3)
|  +- CPU Core #4    :  1397.42  1397.42  1397.42 (/amdcpu/0/clock/4)
|  +- CPU Core #5    :  3992.64  3992.64  3992.64 (/amdcpu/0/clock/5)
|  +- CPU Core #6    :  3992.64  3992.64  3992.64 (/amdcpu/0/clock/6)
|  +- CPU Package    :   51.375   51.375   51.375 (/amdcpu/0/temperature/0)
|  +- CPU Total      :  8.68056  8.68056  8.68056 (/amdcpu/0/load/0)
|  +- CPU Core #1    :     12.5     12.5     12.5 (/amdcpu/0/load/1)
|  +- CPU Core #2    :        0        0        0 (/amdcpu/0/load/2)
|  +- CPU Core #3    :        0        0        0 (/amdcpu/0/load/3)
|  +- CPU Core #4    :  4.16667  4.16667  4.16667 (/amdcpu/0/load/4)
|  +- CPU Core #5    :        0        0        0 (/amdcpu/0/load/5)
|  +- CPU Core #6    :  35.4167  35.4167  35.4167 (/amdcpu/0/load/6)
|  +- CPU Package    :  9.01873  9.01873  9.01873 (/amdcpu/0/power/0)
|  +- CPU Cores      :  3.75506  3.75506  3.75506 (/amdcpu/0/power/1)
|  +- CPU Core #1    : 0.265584 0.265584 0.265584 (/amdcpu/0/power/2)
|  +- CPU Core #2    : 0.0440937 0.0440937 0.0440937 (/amdcpu/0/power/3)
|  +- CPU Core #3    : 0.106635 0.106635 0.106635 (/amdcpu/0/power/4)
|  +- CPU Core #4    : 0.663924 0.663924 0.663924 (/amdcpu/0/power/5)
|  +- CPU Core #5    : 0.181745 0.181745 0.181745 (/amdcpu/0/power/6)
|  +- CPU Core #6    :  2.49308  2.49308  2.49308 (/amdcpu/0/power/7)
|
+- AMD Radeon Graphics (/atigpu/0)
|  +- GPU Core       :    0.001    0.001    0.001 (/atigpu/0/voltage/0)
|  +- GPU Core       :      400      400      400 (/atigpu/0/clock/0)
|  +- GPU Memory     :     1200     1200     1200 (/atigpu/0/clock/1)
|  +- GPU Core       :        1        1        1 (/atigpu/0/load/0)

--------------------------------------------------------------------------------

Parameters

|
+- Lenovo 20UF0008CD (/mainboard)
|
+- AMD Ryzen 5 PRO 4650U (/amdcpu/0)
|  |
|  +- CPU Package (/amdcpu/0/temperature/0)
|  |  +- Offset [°C] : 0 : 0
|
+- AMD Radeon Graphics (/atigpu/0)

--------------------------------------------------------------------------------

Mainboard

SMBIOS Version: 3.2

BIOS Vendor: LENOVO
BIOS Version: R1CET60W(1.29 )

System Manufacturer: LENOVO
System Name: XXXXXXXXXX
System Version: ThinkPad X13 Gen 1

Mainboard Manufacturer: LENOVO
Mainboard Name: XXXXXXXX
Mainboard Version: Not Defined

Processor Manufacturer: Advanced Micro Devices, Inc.
Processor Version: AMD Ryzen 5 PRO 4650U with Radeon Graphics
Processor Core Count: 6
Processor Core Enabled: 6
Processor Thread Count: 12
Processor External Clock: 100 Mhz

Memory Device [0] Manufacturer: Samsung
Memory Device [0] Part Number: M471A1G44AB0-CWE
Memory Device [0] Device Locator: DIMM 0
Memory Device [0] Bank Locator: P0 CHANNEL A
Memory Device [0] Speed: 3200 MHz

Memory Device [1] Manufacturer: Samsung
Memory Device [1] Part Number: M471A1G44AB0-CWE
Memory Device [1] Device Locator: DIMM 0
Memory Device [1] Bank Locator: P0 CHANNEL B
Memory Device [1] Speed: 3200 MHz

```

xud6, 2021-03-15:
OHMR report on Ryzen 5 3600 DIY desktop
```
Open Hardware Monitor Report

--------------------------------------------------------------------------------

Version: 0.9.6.0

--------------------------------------------------------------------------------

Common Language Runtime: 4.0.30319.42000
Operating System: Microsoft Windows NT 6.2.9200.0
Process Type: 32-Bit

--------------------------------------------------------------------------------

Sensors

|
+- MSI B450M MORTAR MAX (MS-7B89) (/mainboard)
|  |
|  +- Nuvoton NCT6797D (/lpc/nct6797d)
|  |  +- CPU VCore      :    1.344    1.344    1.344 (/lpc/nct6797d/voltage/0)
|  |  +- Voltage #2     :    1.008    1.008    1.008 (/lpc/nct6797d/voltage/1)
|  |  +- AVCC           :    3.392    3.392    3.392 (/lpc/nct6797d/voltage/2)
|  |  +- 3VCC           :    3.344    3.344    3.344 (/lpc/nct6797d/voltage/3)
|  |  +- Voltage #5     :    1.016    1.016    1.016 (/lpc/nct6797d/voltage/4)
|  |  +- Voltage #6     :    0.144    0.144    0.144 (/lpc/nct6797d/voltage/5)
|  |  +- Voltage #7     :    0.872    0.872    0.872 (/lpc/nct6797d/voltage/6)
|  |  +- 3VSB           :    3.392    3.392    3.392 (/lpc/nct6797d/voltage/7)
|  |  +- VBAT           :    3.328    3.328    3.328 (/lpc/nct6797d/voltage/8)
|  |  +- VTT            :     1.84     1.84     1.84 (/lpc/nct6797d/voltage/9)
|  |  +- Voltage #12    :    0.792    0.792    0.792 (/lpc/nct6797d/voltage/11)
|  |  +- Voltage #13    :    1.104    1.104    1.104 (/lpc/nct6797d/voltage/12)
|  |  +- Voltage #14    :     0.68     0.68     0.68 (/lpc/nct6797d/voltage/13)
|  |  +- Voltage #15    :    1.544    1.544    1.544 (/lpc/nct6797d/voltage/14)
|  |  +- Temperature #1 :       41       41       41 (/lpc/nct6797d/temperature/1)
|  |  +- Temperature #2 :       30       30       30 (/lpc/nct6797d/temperature/2)
|  |  +- Temperature #3 :     33.5     33.5     33.5 (/lpc/nct6797d/temperature/3)
|  |  +- Temperature #5 :       37       37       37 (/lpc/nct6797d/temperature/5)
|  |  +- Temperature #6 :       -3       -3       -3 (/lpc/nct6797d/temperature/6)
|  |  +- Fan #2         :  1351.35  1351.35  1351.35 (/lpc/nct6797d/fan/1)
|  |  +- Fan #3         :  1195.75  1195.75  1195.75 (/lpc/nct6797d/fan/2)
|  |  +- Fan #4         :  4231.98  4231.98  4231.98 (/lpc/nct6797d/fan/3)
|  |  +- Fan #5         :  1125.94  1125.94  1125.94 (/lpc/nct6797d/fan/4)
|  |  +- Fan Control #1 :  49.8039  49.8039  49.8039 (/lpc/nct6797d/control/0)
|  |  +- Fan Control #2 :  78.8235  78.8235  78.8235 (/lpc/nct6797d/control/1)
|  |  +- Fan Control #3 :  82.3529  82.3529  82.3529 (/lpc/nct6797d/control/2)
|  |  +- Fan Control #4 :  57.6471  57.6471  57.6471 (/lpc/nct6797d/control/3)
|  |  +- Fan Control #5 :  81.1765  81.1765  81.1765 (/lpc/nct6797d/control/4)
|  |  +- Fan Control #6 :  49.8039  49.8039  49.8039 (/lpc/nct6797d/control/5)
|  |  +- Fan Control #7 :  49.8039  49.8039  49.8039 (/lpc/nct6797d/control/6)
|
+- AMD Ryzen 5 3600 (/amdcpu/0)
|  +- Bus Speed      :  100.001  100.001  100.001 (/amdcpu/0/clock/0)
|  +- CPU Core #1    :  4150.04  4150.04  4150.04 (/amdcpu/0/clock/1)
|  +- CPU Core #2    :  4150.04  4150.04  4150.04 (/amdcpu/0/clock/2)
|  +- CPU Core #3    :  4150.04  4150.04  4150.04 (/amdcpu/0/clock/3)
|  +- CPU Core #4    :  3600.03  3600.03  3600.03 (/amdcpu/0/clock/4)
|  +- CPU Core #5    :  3600.03  3600.03  3600.03 (/amdcpu/0/clock/5)
|  +- CPU Core #6    :  3600.03  3600.03  3600.03 (/amdcpu/0/clock/6)
|  +- CPU Package    :    52.25    52.25    52.25 (/amdcpu/0/temperature/0)
|  +- CPU CCD #1     :     58.5     58.5     58.5 (/amdcpu/0/temperature/4)
|  +- CPU Total      :  12.0538  12.0538  12.0538 (/amdcpu/0/load/0)
|  +- CPU Core #1    :  14.8215  14.8215  14.8215 (/amdcpu/0/load/1)
|  +- CPU Core #2    :  13.8839  13.8839  13.8839 (/amdcpu/0/load/2)
|  +- CPU Core #3    :  14.9123  14.9123  14.9123 (/amdcpu/0/load/3)
|  +- CPU Core #4    :  6.89655  6.89655  6.89655 (/amdcpu/0/load/4)
|  +- CPU Core #5    :  6.89655  6.89655  6.89655 (/amdcpu/0/load/5)
|  +- CPU Core #6    :  14.9123  14.9123  14.9123 (/amdcpu/0/load/6)
[... 177 lines cut ...]

Memory Device [0] Manufacturer: Unknown
Memory Device [0] Part Number: Unknown
Memory Device [0] Device Locator: DIMM 0
Memory Device [0] Bank Locator: P0 CHANNEL A
Memory Device [0] Speed: 3600 MHz

Memory Device [1] Manufacturer: A-DATA Technology
Memory Device [1] Part Number: DDR4 3600
Memory Device [1] Device Locator: DIMM 1
Memory Device [1] Bank Locator: P0 CHANNEL A
Memory Device [1] Speed: 3600 MHz

Memory Device [2] Manufacturer: Unknown
Memory Device [2] Part Number: Unknown
Memory Device [2] Device Locator: DIMM 0
Memory Device [2] Bank Locator: P0 CHANNEL B
Memory Device [2] Speed: 3600 MHz

Memory Device [3] Manufacturer: A-DATA Technology
Memory Device [3] Part Number: DDR4 3600
Memory Device [3] Device Locator: DIMM 1
Memory Device [3] Bank Locator: P0 CHANNEL B
Memory Device [3] Speed: 3600 MHz
```

gecon, 2021-12-15:
```
# sensors
k10temp-pci-00c3
Adapter: PCI adapter
Vcore:        +1.36 V
Vsoc:         +1.01 V
Tctl:         +70.8°C
Tdie:         +70.8°C
Tccd1:        +71.5°C
Tccd2:        +68.8°C
Icore:       +57.00 A
Isoc:         +6.50 A

nvme-pci-0900
Adapter: PCI adapter
Composite:    +33.9°C  (low  = -273.1°C, high = +79.8°C)
                       (crit = +82.8°C)
Sensor 1:     +33.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +41.9°C  (low  = -273.1°C, high = +65261.8°C)

nvme-pci-0800
Adapter: PCI adapter
Composite:    +32.9°C  (low  = -273.1°C, high = +79.8°C)
                       (crit = +82.8°C)
Sensor 1:     +32.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +40.9°C  (low  = -273.1°C, high = +65261.8°C)
```

on:

```
# lscpu
Architecture:        x86_64
CPU op-mode(s):      32-bit, 64-bit
Byte Order:          Little Endian
CPU(s):              24
On-line CPU(s) list: 0-23
Thread(s) per core:  2
Core(s) per socket:  12
Socket(s):           1
NUMA node(s):        1
Vendor ID:           AuthenticAMD
BIOS Vendor ID:      Advanced Micro Devices, Inc.
CPU family:          23
Model:               113
Model name:          AMD Ryzen 9 3900 12-Core Processor
BIOS Model name:     AMD Ryzen 9 3900 12-Core Processor
Stepping:            0
CPU MHz:             3786.165
CPU max MHz:         3100.0000
CPU min MHz:         2200.0000
BogoMIPS:            6188.24
Virtualization:      AMD-V
L1d cache:           32K
L1i cache:           32K
L2 cache:            512K
L3 cache:            16384K
NUMA node0 CPU(s):   0-23
Flags:               fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_good nopl nonstop_tsc cpuid extd_apicid aperfmperf pni pclmulqdq monitor ssse3 fma cx16 sse4_1 sse4_2 movbe popcnt aes xsave avx f16c rdrand lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw ibs skinit wdt tce topoext perfctr_core perfctr_nb bpext perfctr_llc mwaitx cpb cat_l3 cdp_l3 hw_pstate sme ssbd mba sev ibpb stibp vmmcall fsgsbase bmi1 avx2 smep bmi2 cqm rdt_a rdseed adx smap clflushopt clwb sha_ni xsaveopt xsavec xgetbv1 xsaves cqm_llc cqm_occup_llc cqm_mbm_total cqm_mbm_local nt_good clzero irperf xsaveerptr rdpru wbnoinvd arat npt lbrv svm_lock nrip_save tsc_scale vmcb_clean flushbyasid decodeassists pausefilter pfthreshold avic v_vmsave_vmload vgif umip rdpid overflow_recov succor smca
```

and
```
# sensors -u
k10temp-pci-00c3
Adapter: PCI adapter
Vcore:
  in0_input: 1.400
Vsoc:
  in1_input: 1.013
Tctl:
  temp1_input: 63.125
Tdie:
  temp2_input: 63.125
Tccd1:
  temp3_input: 52.250
Tccd2:
  temp4_input: 49.750
Icore:
  curr1_input: 41.000
Isoc:
  curr2_input: 6.500

nvme-pci-0900
Adapter: PCI adapter
Composite:
  temp1_input: 33.850
  temp1_max: 79.850
  temp1_min: -273.150
  temp1_crit: 82.850
  temp1_alarm: 0.000
Sensor 1:
  temp2_input: 33.850
  temp2_max: 65261.850
  temp2_min: -273.150
Sensor 2:
  temp3_input: 41.850
  temp3_max: 65261.850
  temp3_min: -273.150

nvme-pci-0800
Adapter: PCI adapter
Composite:
  temp1_input: 32.850
  temp1_max: 79.850
  temp1_min: -273.150
  temp1_crit: 82.850
  temp1_alarm: 0.000
Sensor 1:
  temp2_input: 32.850
  temp2_max: 65261.850
  temp2_min: -273.150
Sensor 2:
  temp3_input: 40.850
  temp3_max: 65261.850
  temp3_min: -273.150

```