nobody43/zabbix-mini-IPMI #3: max CPU temperature issue

gatsu1984, 2017-07-05:
hello,
trying to use your zabbix-mini-IMPI, but getting error on Maximum CPU Temperature, and as i understand since it cant get max cpu temp, it also not getting other info by sender.
This is from server log:
"example.org:temp.cpu[max]" changed: Received value [zabbix_sender.exe [376]: invalid parameter ""44] is not suitable for value type [Numeric (unsigned)] and data type [Decimal].
So it shows max temp, but with it also showing some sender error about invalid parameters and not working couse of this.

root@zabbixserver:~# zabbix_get -s 192.168.1.15 -k temp.cpu[max]
zabbix_sender.exe [1592]: invalid parameter ""
45

can you help me somehow?
thx

nobodysu, 2017-07-05:
Good day.
What about your userparameter? No spaces there? Quoting must be a proper way:
`UserParameter=temp.disk[*], python "c:\zabbix-agent\scripts\temp-disk.py" "$1"`

gatsu1984, 2017-07-05:
userparameters exact as in your file. i just copied it.  

temp.disk is working normall, only zabbixsender working strange in temp.cpu

nobodysu, 2017-07-05:
Please replace the line that I provided (quote the arguments) and restart the agent.
If that doesn't help try using 32bit sender.

gatsu1984, 2017-07-05:
same.

with 32 bit sender

 29515:20170705:174639.161 item "example.org:temp.cpu[max]" became not supported: Received value [zabbix_sender.exe [1676]: invalid parameter ""4] is not suitable for value type [Numeric (unsigned)] and data type [Decimal]

nobodysu, 2017-07-05:
Please attach your configuration from zabbix-ohmr-wrapper.py (header) and whole zabbix config file.

gatsu1984, 2017-07-05:
[ohmr-wrapper.txt](https://github.com/nobodysu/zabbix-mini-IPMI/files/1125158/ohmr-wrapper.txt)

[zabbix_agentd.txt](https://github.com/nobodysu/zabbix-mini-IPMI/files/1125162/zabbix_agentd.txt)

nobodysu, 2017-07-05:
Try changing line 116 of ohmr-wrapper to this:
`senderProc = subprocess.Popen([senderPath, '-c', agentConf, '-i', '-'], stdin=subprocess.PIPE, stdout=subprocess.DEVNULL, universal_newlines=True)`
(remove isDebug)

gatsu1984, 2017-07-05:
after changing temp.cpu[max] working normal, but not getting other items

nobodysu, 2017-07-05:
`zabbix_agentd.exe -V`
`zabbix_sender.exe -V`
?

gatsu1984, 2017-07-05:
c:\zabbix\bin\win64>zabbix_agentd.exe -V
zabbix_agentd Win64 (service) (Zabbix) 3.2.0
Revision 62444 13 September 2016, compilation time: Sep 12 2016 15:28:10

Copyright (C) 2016 Zabbix SIA
License GPLv2+: GNU GPL version 2 or later <http://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it according to
the license. There is NO WARRANTY, to the extent permitted by law.

c:\zabbix\bin\win64>zabbix_sender.exe -V
zabbix_sender (Zabbix) 3.2.0
Revision 62444 13 September 2016, compilation time: Sep 12 2016 15:28:10

Copyright (C) 2016 Zabbix SIA
License GPLv2+: GNU GPL version 2 or later <http://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it according to
the license. There is NO WARRANTY, to the extent permitted by law.

gatsu1984, 2017-07-05:
on zabbix server

root@zabbixserver:/etc/zabbix/scripts# zabbix_agentd -V
zabbix_agentd (daemon) (Zabbix) 3.2.6
Revision 67849 4 May 2017, compilation time: May  6 2017 01:09:33

Copyright (C) 2017 Zabbix SIA
License GPLv2+: GNU GPL version 2 or later <http://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it according to
the license. There is NO WARRANTY, to the extent permitted by law.

nobodysu, 2017-07-05:
Also `python -V` please.

gatsu1984, 2017-07-05:
windows
c:\zabbix\bin\win64>python -V
Python 3.6.1

but also have computers with 3.4 that working same as this

linux
root@zabbixserver:/etc/zabbix/scripts# python -V
Python 2.7.12

nobodysu, 2017-07-07:
Unfortunately I can't reproduce this behavior with 3.2 agent/sender. Are you sure that `vlt` and `fan` is present on machine and still missing (beside present `temp.cpu*`)?

gatsu1984, 2017-07-07:
yes im sure that its present, becouse i can get it if i wil run python script by hand with "-v" 

since i dont know python i fixed it in this way

i didnt delete isDebug and ad this in senderProc : stderr=subprocess.DEVNULL

senderProc = subprocess.Popen([senderPath, isDebug, '-c', agentConf, '-i', '-'], stdin=subprocess.PIPE, stdout=subprocess.DEVNULL, stderr=subprocess.DEVNULL, universal_newlines=True)

and put this in agent config so it cut all except temp max after runing script with -v

UserParameter=temp.cpu[*],for /F "eol=- skip=19" %a in ('c:\zabbix\scripts\zabbix-ohmr-wrapper.py -v') do @echo %a

this way all items sending for me

nobodysu, 2017-07-07:
Glad you sorted it out. I planned to rewrite this script in more convenient, LLD-way in a couple of months. Most likely it will be fixed there.

gatsu1984, 2017-07-07:
good to hear it :)

nobodysu, 2017-11-06:
Fixed in v2.

closed 2017-11-06 by nobodysu