nobody43/zabbix-mini-IPMI #4: zabbix_sender was not found!

mausspb, 2017-09-27:
OS Linux, i get this error, when i'm testing you scripts. 
Zabbix agent - v.3.4.2
Zabbix server - v.3.4.2

`# ./zabbix-lmsensors-wrapper.py -v ` (and other testing command return error with path zabbix_sender ).

> ./zabbix-lmsensors-wrapper.py -v
> 
>   ServerActive config value:
> 127.0.0.1
> 
>   Hostname config value:
> Zabbix server
> 
>   zabbix_sender path:
> zabbix_sender
> was not found!
> 

--

`# zabbix_get -s 127.0.0.1 -k temp.cpu[max]`

```
ZBX_NOTSUPPORTED: Traceback (most recent call last):
  File "/etc/zabbix/scripts/zabbix-lmsensors-wrapper.py", line 106, in <module>
    senderProc = subprocess.Popen([senderPath, isDebug, '-c', agentConf, '-i', '-'], stdin=subprocess.PIPE, stdout=subprocess.DEVNULL, universal_newlines=True)
  File "/usr/lib/python3.5/subprocess.py", line 947, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.5/subprocess.py", line 1551, in _execute_child
    raise child_exception_type(errno_num, err_msg)
FileNotFoundError: [Errno 2] No such file or directory: 'zabbix_sender'
root@reder-netTop:/etc/zabbix/scripts# zabbix_get -s 127.0.0.1 -k temp.cpu[max]
ZBX_NOTSUPPORTED: Traceback (most recent call last):
  File "/etc/zabbix/scripts/zabbix-lmsensors-wrapper.py", line 106, in <module>
    senderProc = subprocess.Popen([senderPath, isDebug, '-c', agentConf, '-i', '-'], stdin=subprocess.PIPE, stdout=subprocess.DEVNULL, universal_newlines=True)
  File "/usr/lib/python3.5/subprocess.py", line 947, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.5/subprocess.py", line 1551, in _execute_child
    raise child_exception_type(errno_num, err_msg)
FileNotFoundError: [Errno 2] No such file or directory: 'zabbix_sender'
```

 **How fix it ?**

nobodysu, 2017-09-27:
`which zabbix_sender`?
You can install it with `apt-get install zabbix-sender` / `yum install zabbix-sender`
Also please take notice:
https://github.com/nobodysu/zabbix-mini-IPMI/issues/3

mausspb, 2017-09-28:
This is solved !
But i get next error 
```
zabbix_sender [25495]: DEBUG: send value error: cannot connect to [[127.0.0.1]:10051]: [111] Connection refused
43
```

zabbix_sender want connect to 127.0.0.1:10051, what for ( it is no server address ) ?

nobodysu, 2017-09-28:
You must configure your zabbix agent (/etc/zabbix/zabbix_agentd.conf). Provide `Server`, `ServerActive` and `Hostname`. Increase `Timeout`.