nobody43/zabbix-mini-IPMI #7: Python not recognized?

therealfusionz, 2017-11-20:
Hey
First of all, I'm not really familiar with stuff like this(scripts etc), but I am trying to get hardware temps on Windows(10) agent. 
zabbix_agentd.log:
Requested [mini.cputemp.discovery[get,Win10]]
  9008:20171120:141458.258 'python' is not recognized as an internal or external command, operable program or batch file.

Cannot run mini_ipmi_ohmr.py or mini_ipmi_smartctl.py on CMD window aswell.
mini_ipmi_smartctl.py: Not supported. Use 'get' or 'getverb'.

C:\zabbix>python scripts/mini_ipmi_ohmr.py -v
Traceback (most recent call last):
  File "scripts/mini_ipmi_ohmr.py", line 286, in <module>
    host = '"' + sys.argv[2] + '"'
IndexError: list index out of range

Using Python 3.6.3 on Windows.

nobodysu, 2017-11-20:
First - make sure you are using the latest version from github.
With python you have two options:
- add it to PATH (during installation is simpler)
- specify python executable manually in `userparameter_mini-ipmi2.conf`:
```UserParameter=mini.disktemp.discovery[*], "C:\Python36\python.exe" "C:\zabbix-agent\scripts\mini_ipmi_smartctl.py" "$1" "$2"
UserParameter=mini.cputemp.discovery[*], "C:\Python36\python.exe" "C:\zabbix-agent\scripts\mini_ipmi_ohmr.py" "$1" "$2"```

therealfusionz, 2017-11-21:
Redownloaded the latest version, reconfigured and managed to get a some kind of response. Without any sent/recieved data, but at least a response.
Log:
https://pastebin.com/CyagMgX0

nobodysu, 2017-11-21:
Its not working now? After 22 minutes it should.
If not - execute on server:
```
# apt-get install zabbix-get
$ zabbix_get -s Win10 -k mini.cputemp.discovery[get,"Win10"]
$ zabbix_get -s Win10 -k mini.cputemp.discovery[getverb,"Win10"]
```
and post output of last two.

therealfusionz, 2017-11-21:
**getverb:**
Server=xxxx
Could not find 'ServerActive' setting in config!
Timeout=30

  Please report any issues to:
https://github.com/nobodysu/zabbix-mini-IPMI/issues
zabbix_sender.exe [1792]: ERROR: 'ServerActive' parameter required
Sending failed.

  Note: the sender will fail if server did not gather LLD previously.

  Data sent to zabbix sender:

"Win10" mini.info[OHMRver] "0.8.0.2"
"Win10" mini.brd.info[SMBIOSversion] "2.7"
"Win10" mini.brd.info[BIOSvendor] "LENOVO"
"Win10" mini.brd.info[BIOSversion] "G3ETA7WW(2.67)"
"Win10" mini.brd.info[MainboardManufacturer] "LENOVO"
"Win10" mini.brd.info[MainboardName] "246835G"
"Win10" mini.brd.info[MainboardVersion] "Not Defined"
"Win10" mini.cpu.info[ConfigStatus] "NOGPUS"
"Win10" mini.cpu.info[cpu0,ID] "Intel Core i5-3210M"
"Win10" mini.cpu.info[cpu0,TjMax] "105"
"Win10" mini.cpu.info[cpu0,CPUstatus] "PROCESSED"
"Win10" mini.cpu.temp[cpu0,core0] "57"
"Win10" mini.cpu.temp[cpu0,core1] "52"
"Win10" mini.cpu.temp[cpu0,MAX] "57"
"Win10" mini.cpu.temp[MAX] "57"

Tried changing the Server=xxxx parameter to ServerActive=xxxx:10050(and without the port) in agent config but Zabbix Agent service won't start up saying it didn't respond in timely fashion.

nobodysu, 2017-11-21:
It told you:
`Could not find 'ServerActive' setting in config!`
Set it in `zabbix_agentd.conf` to your zabbix server.

therealfusionz, 2017-11-21:
ServerActive parameter was already set in Zabbix server, tried chaning and restarting the services as well.

nobodysu, 2017-11-21:
Server or agent? You need ServerActive on agent:
```zabbix_sender.exe [1792]: ERROR: 'ServerActive' parameter required```

therealfusionz, 2017-11-21:
On Zabbix server, it was already preconfigured, but on agent machine, the Zabbix Agent service won't start if I add ServerActive=IP:10050 or just IP. I tried removing Server= parameter aswell.

nobodysu, 2017-11-21:
You need both `Server=192.0.2.1` and `ServerActive=192.0.2.1` set and uncommented in `zabbix_agentd.conf` for it to work. (IP is an example)

nobodysu, 2017-11-21:
Maybe you should start with new config file, from the distributive?

therealfusionz, 2017-11-21:
Now it has started active check, but fails because it cannot connect to the server(port specifically). Port in ServerActive parameter should be 10050, right? I've also tried 10051 and even without specifying the port, but log says it cannot connect to the server.
Server itself says "Warning: incorrect answer from server".

nobodysu, 2017-11-21:
Drop the port completely, its for special cases, use only IP.
Error could indicate firewall or selinux restrictions on server.

therealfusionz, 2017-11-21:
Alright, finally got it working!

Not sure what was wrong but I ended up redownloading and reconfiguring both zabbix agent for windows and zabbix-mini-IPMI packages. 

Also enabled port 10050
_iptables -I INPUT -p tcp --dport 10050 -j ACCEPT_

No idea why this wasn't working before I reconfigured everything from the scratch.
A big thank you for helping out a newb!

nobodysu, 2017-11-21:
Great!

closed 2017-11-21 by nobodysu