nobody43/zabbix-mini-IPMI #9: Help with error needed unprocessed items.

imcookie, 2018-01-15:
Here is my part of log file. How exactly i can know why 4 items failed? I didnt find any errors or warnings in zabbix_agent log file. 

zabbix_sender [4504]: DEBUG: answer [{"response":"success","info":"processed: 2; failed: 4; total: 6; seconds spent: 0.000086"}]
info from server: "processed: 2; failed: 4; total: 6; seconds spent: 0.000086"
sent: 6; skipped: 0; total: 6

nobodysu, 2018-01-15:
Zabbix agent does not provide that functionality. 
The only way you could figure that out is by using getverb parameter with the script and comparing recieved data in host's 'Latest data'.

imcookie, 2018-01-15:
Here what i have.
"Mail" mini.disk.temp[sda_sat_megaraid_4] 27
"Mail" mini.disk.info[sda_sat_megaraid_4,DriveStatus] "PROCESSED"
"Mail" mini.disk.temp[sda_sat_megaraid_5] 30
"Mail" mini.disk.info[sda_sat_megaraid_5,DriveStatus] "PROCESSED"
"Mail" mini.disk.temp[MAX] 30

and in latest data i have only

DISK temperature configuration status | 16.01.2018 00:50:53 | CONFIGURED
and
All DISKs maximum temperature | 16.01.2018 00:50:53 | 30 C

and another strange thing, zabbix-server dosnt retrive this data automatically, i get it only by executing   zabbix_get -s mail  -k mini.disktemp.discovery[getverb,"Mail"]

imcookie, 2018-01-15:
Can you please provide example to manually way to execute mini_ipmi_smartctl.py. Wich paramaters i need to pas to mini_ipmi_smartctl.py?

nobodysu, 2018-01-15:
Zabbix server must first discover available items (LLD) - thats why sanding failed.
What about
```zabbix_get -s mail -k mini.disktemp.discovery[get,"Mail"]```
Any errors?
_Have you waited for 22 minutes for discovery?_

nobodysu, 2018-01-15:
You can execute it locally like this:
```./mini_ipmi_smartctl.py get 'Example host'```
But you don't need to - examples in [testing](https://github.com/nobodysu/zabbix-mini-IPMI#testing) are perfectly suitable for your needs - they cover connectivity too.

imcookie, 2018-01-15:
info from server: "processed: 0; failed: 1; total: 1; seconds spent: 0.000029"
sent: 1; skipped: 0; total: 1
{
    "data": [
        {
            "{#DISK}": "sda_sat_megaraid_4"
        }, 
        {
            "{#DISK}": "sda_sat_megaraid_5"
        }
    ]
}
I waited like 20 hours ... Or probably i executed zabbix_get before (LLD) and it's the reason?  
On another server everything working well.

nobodysu, 2018-01-15:
Are you receiving clean LLD or with top message (info from server)?
Your next bet is to check `Hosts - Mail - Discovery rules - DISK temperature discovery - Info` section for errors (on the right).
Does another server have megaraid?

imcookie, 2018-01-15:
WIth top info.
root@zabbix:~# zabbix_get -s mail  -k mini.disktemp.discovery[get,"Mail"]
info from server: "processed: 0; failed: 1; total: 1; seconds spent: 0.000028"
sent: 1; skipped: 0; total: 1
{
    "data": [
        {
            "{#DISK}": "sda_sat_megaraid_4"
        }, 
        {
            "{#DISK}": "sda_sat_megaraid_5"
        }
    ]
}
root@zabbix:~# 
[image]
Another servers also with megaraid. But debian versions different. (wheezy - mail. jessie - another one).

imcookie, 2018-01-15:
I recreated host in zabbix, assigned mini_ipmi template again and waited for 22 mins. Now in discovery rules i got error (Value should be a JSON object.)

nobodysu, 2018-01-15:
Please provide agent, sender and python exact versions. Also copy your script configuration here.

imcookie, 2018-01-16:
zabbix_agentd (daemon) (Zabbix) 3.0.14
zabbix_sender (Zabbix) 3.0.14
Python 3.2.3

`#!/usr/bin/env python3

## Installation instructions: https://github.com/nobodysu/zabbix-mini-IPMI ##

checkStandby = 'no'   # whether to check disks in STANDBY mode or not, 'yes' or 'no'
                      # if 'Update interval' is less than OS setting for STANDBY disks - it will never enter this state

ctlPath = r'smartctl'
#ctlPath = r'C:\Program Files\smartmontools\bin\smartctl.exe'   # if smartctl isn't in PATH
#ctlPath = r'/usr/local/sbin/smartctl'

# path to second send script
senderPyPath = r'/etc/zabbix/scripts/sender_wrapper.py'             # Linux
#senderPyPath = r'C:\zabbix-agent\scripts\sender_wrapper.py'        # Win
#senderPyPath = r'/usr/local/etc/zabbix/scripts/sender_wrapper.py'  # BSD

# path to zabbix agent configuration file
agentConf = r'/etc/zabbix/zabbix_agentd.conf'                       # Linux
#agentConf = r'C:\zabbix_agentd.conf'                               # Win
#agentConf = r'/usr/local/etc/zabbix3/zabbix_agentd.conf'           # BSD

senderPath = r'/usr/bin/zabbix_sender'                                       # Linux, BSD
#senderPath = r'C:\zabbix-agent\bin\win32\zabbix_sender.exe'        # Win

timeout = '80'   # how long the script must wait between LLD and sending, increase if data received late (does not affect windows)
                 # this setting MUST be lower than 'Update interval' in discovery rule

# manually provide disk list or RAID configuration if needed
#diskListManual = []
# like this:
diskListManual = ['/dev/sda -d sat+megaraid,4', '/dev/sda -d sat+megaraid,5']
# more info: https://www.smartmontools.org/wiki/Supported_RAID-Controllers

## End of configuration ##

import sys
import subprocess
import re
from shlex import split
from sender_wrapper import (readConfig, processData, replaceStr)

def listDisks():
    '''Determines available disks. Can be skipped.'''
    try:
        p = subprocess.check_output([ctlPath, '--scan'], universal_newlines=True)   # scan the disks
        error = ''
    except OSError as e:
        p = ''

        if e.args[0] == 2:
            error = 'SCAN_OS_NOCMD'
        else:
            error = 'SCAN_OS_ERROR'
    except Exception as e:
        try:   # extra safe
            p = e.output
        except:
            p = ''
        error = 'SCAN_UNKNOWN_ERROR'
        if sys.argv[1] == 'getverb':
            raise

    disks = re.findall(r'^(/dev/[^ ]+)', p, re.M)   # determine full device names

    return error, disks

[... 132 lines cut ...]
        scanConfigError = ''
        diskList = diskListManual   # or just use manually provided settings

    if diskList:
        getDisksTempSCT_Out = getDisksTempSCT()

        diskConfigError = getDisksTempSCT_Out[0]   # D_OS_NOCMD, D_OS_ERROR, NOTEMPS, UNKNOWN_EXC_ERROR
        senderData = getDisksTempSCT_Out[1]
    else:
        diskConfigError = 'NODISKS'
        senderData = []

    if scanConfigError != '':
        senderData.append(host + ' mini.disk.info[ConfigStatus] "' + scanConfigError + '"')   # takes precedence
    elif diskConfigError != '':
        senderData.append(host + ' mini.disk.info[ConfigStatus] "' + diskConfigError + '"')   # on mixed errors
    else:
        senderData.append(host + ' mini.disk.info[ConfigStatus] "CONFIGURED"')   # signals that client host is configured

    for d in diskList:
        dR = replaceStr(d)
        jsonData.append({'{#DISK}':dR})   # available disks must always be populated to LLD

    processData(senderData, jsonData, agentConf, senderPyPath, senderPath, timeout, host)
`

imcookie, 2018-01-16:
Upd
I found this in my zabbix-server.log:
` 56318:20180116:220322.670 item "Mail:mini.disktemp.discovery[get,{HOST.HOST}]" became not supported: Timeout while executing a shell script.`

imcookie, 2018-01-16:
I installed python3.4 look's like it fixed a prolem.

closed 2018-01-16 by imcookie

nobodysu, 2018-01-17:
Yeah, it seems older python versions are not as compatible as 3.4. I still need to fix it, thank you for the report.