nobody43/zabbix-mini-IPMI #10: ZBX_NOTSUPPORTED: Timeout while executing a shell script.

leandroz3ta, 2018-02-07:
Config workstation: 
Intel Pentium G2020 2.9 Ghz
6 GB RAM
Windows 7 Professional 64 bits
When running the command zabbix_get -s 192.168.1.193 -k mini.cputemp.discovery [get, "Host example"]
The Zabbix Server returns Time Out, although I can not see any task manager that is OpenHardwareMonitor.exe is executed by the script.
Note: In the mini_ipmi_ohmr file reference is made to OpenHardwareMonitorReport.exe, however the version I downloaded does not have it but OpenHardwareMonitor.exe in which version is this OpenHardwareMonitorReport.exe executable?

leandroz3ta, 2018-02-07:
Thank you! I was able to fix it, I checked it here and really have to have this executable.

closed 2018-02-07 by leandroz3ta

nobodysu, 2018-02-07:
Great!