nobody43/zabbix-mini-IPMI #12: Integrated CardReader causes discovery to fail

jschfr, 2018-04-14:
Hi,
I have an integrated card reader in the windows client machine I want to monitor with mini-IPMI. Explorer shows me 4 devices (E:, F:, G:, H:,). When testing mini-IPMI through `zabbix_get` on the same machine I get the following 

```
PS C:\Program Files\Zabbix Agent> ./zabbix_get -s 127.0.0.1 -k mini.disktemp.discovery[get,myPC]
Traceback (most recent call last):
  File "C:\Program Files\Zabbix Agent\scripts\mini_ipmi_smartctl.py", line 208, in <module>
    getDisksTempSCT_Out = getDisksTempSCT()
  File "C:\Program Files\Zabbix Agent\scripts\mini_ipmi_smartctl.py", line 146, in getDisksTempSCT
    if e.args[0] == 1 or e.args[0] == 2:   # non-fatal disk error codes are not a concern for temperature monitoring script
IndexError: tuple index out of range
```
after deactivating the card reader through Device-Manager the upper command runs normally. This might be timeout problem, because
```
PS C:\Program Files\smartmontools\bin> .\smartctl.exe --scan
/dev/sda -d ata # /dev/sda, ATA device
/dev/sdb -d ata # /dev/sdb, ATA device
/dev/sdc -d scsi # /dev/sdc, SCSI device
/dev/sdd -d scsi # /dev/sdd, SCSI device
/dev/sde -d scsi # /dev/sde, SCSI device
/dev/sdf -d scsi # /dev/sdf, SCSI device
```
takes about 1-2 seconds to run. But `./zabbix_get -s 127.0.0.1 -k mini.disktemp.discovery[get,myPC]` seems to quit with error message before that time has passed.
After deactivating the card reader through Device-Manager the following runs much faster:
```
PS C:\Program Files\smartmontools\bin> .\smartctl.exe --scan
/dev/sda -d ata # /dev/sda, ATA device
/dev/sdb -d ata # /dev/sdb, ATA device
```

nobodysu, 2018-04-14:
Thanks for the report - I'll investigate it within a week. Please provide error for getverb in a meantime.

jschfr, 2018-04-14:
Test with card reader ACTIVATED:
```
PS C:\Program Files\Zabbix Agent> ./zabbix_get -s 127.0.0.1 -k mini.disktemp.discovery[getverb
,myPC]
Traceback (most recent call last):
  File "C:\Program Files\Zabbix Agent\scripts\mini_ipmi_smartctl.py", line 208, in <module>
    getDisksTempSCT_Out = getDisksTempSCT()
  File "C:\Program Files\Zabbix Agent\scripts\mini_ipmi_smartctl.py", line 146, in getDisksTempSCT
    if e.args[0] == 1 or e.args[0] == 2:   # non-fatal disk error codes are not a concern for temperature monitoring script
IndexError: tuple index out of range
```
Test with card reader DEactivated:
```
PS C:\Program Files\Zabbix Agent> ./zabbix_get -s 127.0.0.1 -k mini.disktemp.discovery[getverb
,myPC]
  Config's main settings:
Server=192.168.0.201, 127.0.0.1
ServerActive=192.168.0.201,127.0.0.1
Timeout=10
()
  Please report any issues to:
https://github.com/nobodysu/zabbix-mini-IPMI/issues
zabbix_sender.exe [320]: DEBUG: answer [{"response":"success","info":"processed: 2; failed: 4; total: 6; seconds spent: 0.000108"}]
info from server: "processed: 2; failed: 4; total: 6; seconds spent: 0.000108"
sent: 6; skipped: 0; total: 6

  Note: the sender will fail if server did not gather LLD previously.

  Data sent to zabbix sender:

"myPC" mini.disk.temp[sda] 0
"myPC" mini.disk.info[sda,DriveStatus] "PROCESSED"
"myPC" mini.disk.temp[sdb] 42
"myPC" mini.disk.info[sdb,DriveStatus] "PROCESSED"
"myPC" mini.disk.temp[MAX] 42
"myPC" mini.disk.info[ConfigStatus] "CONFIGURED"
```

nobodysu, 2018-04-19:
Please try `smartctl -a /dev/sdc` on local machine with cardreader enabled, and post output here.

bobbolous, 2018-04-20:
As requested:
```
smartctl -a /dev/sdc
smartctl 6.6 2017-11-05 r4594 [x86_64-w64-mingw32-w10-1709] (sf-6.6-1)
Copyright (C) 2002-17, Bruce Allen, Christian Franke, www.smartmontools.org

/dev/sdc: Unknown USB bridge [0x058f:0x6362]
Please specify device type with the -d option.

Use smartctl -h to get a usage summary
```

And this:
```
smartctl -a /dev/sdc -d ata
smartctl 6.6 2017-11-05 r4594 [x86_64-w64-mingw32-w10-1709] (sf-6.6-1)
Copyright (C) 2002-17, Bruce Allen, Christian Franke, www.smartmontools.org

Warning: Limited functionality due to missing admin rights
=== START OF INFORMATION SECTION ===
Device Model:     Generic-SD/MMC
Serial Number:    058F63626470
Firmware Version: 1.00
Device is:        Not in smartctl database [for details use: -P showall]
ATA Version is:   [No Information Found]
Local Time is:    Fri Apr 20 09:56:42 2018 MS
SMART support is: Available - device has SMART capability.
                  Enabled status cached by OS, trying SMART RETURN STATUS cmd.
SMART support is: Disabled

SMART Disabled. Use option -s with argument 'on' to enable it.
(override with '-T permissive' option)
```

When trying to enable SMART:
When trying to enable SMART (as Admin):
```
smartctl -a /dev/sdc -d ata -s on
smartctl 6.6 2017-11-05 r4594 [x86_64-w64-mingw32-w10-1709] (sf-6.6-1)
Copyright (C) 2002-17, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Device Model:     Generic-SD/MMC
Serial Number:    058F63626470
Firmware Version: 1.00
Device is:        Not in smartctl database [for details use: -P showall]
ATA Version is:   [No Information Found]
Local Time is:    Fri Apr 20 10:01:22 2018 MS
SMART support is: Available - device has SMART capability.
                  Enabled status cached by OS, trying SMART RETURN STATUS cmd.
SMART support is: Disabled

=== START OF ENABLE/DISABLE COMMANDS SECTION ===
SMART Enabled.
Read SMART Data failed: Function not implemented

=== START OF READ SMART DATA SECTION ===
SMART Status not supported: Function not implemented
SMART overall-health self-assessment test result: UNKNOWN!
SMART Status, Attributes and Thresholds cannot be read.

SMART Error Log not supported

SMART Self-test Log not supported

Selective Self-tests/Logging not supported
```

nobodysu, 2018-04-20:
I can't reproduce the error so please test the fix yourself:
[mini_ipmi_smartctl.zip](https://github.com/nobodysu/zabbix-mini-IPMI/files/1932038/mini_ipmi_smartctl.zip)

bobbolous, 2018-04-20:
Thanks! I will try it next week.

nobodysu, 2018-04-20:
Once you'll get to it, please post your `python -V`.

bobbolous, 2018-04-30:
As requested:
```
python -V
Python 2.7.14
```

but:
```
py -V
Python 3.6.5
```

I now did a complete reinstall of python 3.6.5 without python 2.7, now it works with the 'old' version of `mini_ipmi_smartctl.py` (not the one you provided above). Sorry I did bother you and thanks for your kind help.

closed 2018-04-30 by bobbolous

nobodysu, 2018-04-30:
Please test new version also, with `Unlink and clear` on the host.

bobbolous, 2018-04-30:
Looks good and works for me. I cannot spot a difference to the old version.

nobodysu, 2018-04-30:
Thats good.