nobody43/zabbix-mini-IPMI #13: Incorrect parsing smartctl data for RAID controller

Huron, 2018-04-21:
I have 2 disks on bus 4

```
root@radius:/etc/zabbix/scripts# smartctl --scan
/dev/sda -d scsi # /dev/sda, SCSI device
/dev/bus/4 -d megaraid,0 # /dev/bus/4 [megaraid_disk_00], SCSI device
/dev/bus/4 -d megaraid,1 # /dev/bus/4 [megaraid_disk_01], SCSI device
```
But the discovery script gives me 2 equal entries bus 4 without additional mark of disk number on the bus, so zabbix server cannot to create data items on the host.
```
root@zabbix:~# zabbix_get -s 10.9.2.8 -k "mini.disktemp.discovery[get,radius.lutacom.net]"
{
    "data": [
        {
            "{#DISK}": "sda"
        },
        {
            "{#DISK}": "bus_4"
        },
        {
            "{#DISK}": "bus_4"
        }
    ]
}
```

nobodysu, 2018-04-22:
So unusual, whats the hardware?
I'll fix this within a week.

Huron, 2018-04-23:
```
root@radius:~# lshw -class storage
  *-storage
       description: RAID bus controller
       product: MegaRAID SAS 2108 [Liberator]
       vendor: LSI Logic / Symbios Logic
       physical id: 0
       bus info: pci@0000:08:00.0
       logical name: scsi4
       version: 05
       width: 64 bits
       clock: 33MHz
       capabilities: storage pm pciexpress vpd msi msix bus_master cap_list rom
       configuration: driver=megaraid_sas latency=0
       resources: irq:16 ioport:dc00(size=256) memory:de9bc000-de9bffff memory:de9c0000-de9fffff memory:de900000-de93ffff
...
```

nobodysu, 2018-04-28:
Turns out this fix requires a major rewrite. 
Originally, its intended to manually specify RAID configuration settings, but it would be useful to have this feature. For now, you can set
`diskListManual = ['/dev/bus/4 -d megaraid,0', '/dev/bus/4 -d megaraid,1']`
in [mini_ipmi_smartctl.py
](https://github.com/nobodysu/zabbix-mini-IPMI/blob/master/mini_ipmi_smartctl.py#L29)

commit d8aea9fa (2019-03-11): Refactoring and nvme (#33)

* refactoring and nvme support

* fixes #13, closes #25

* full license text
closed 2019-03-11 by nobodysu