nobody43/zabbix-mini-IPMI #32: Value should be a JSON object.

CheckB, 2019-03-02:
[image]

This project has a similar problem https://github.com/nobodysu/zabbix-smartmontools
p.s. Zabbix 3.0

nobodysu, 2019-03-02:
Its a problem of misconfiguration. Please provide a [getverb](https://github.com/nobodysu/zabbix-mini-IPMI#testing) output.

CheckB, 2019-03-02:
@nobodysu 
Right. I forgot my rights.

New problem (i have 2 CPU):
[image]

nobodysu, 2019-03-02:
Its a cosmetic error - it should not prevent regular script usage. Will be fixed with ohmr script refactoring.
Related: https://github.com/nobodysu/zabbix-mini-IPMI/issues/18

CheckB, 2019-03-02:
How to make smart check work?
This, for example, works correctly: https://github.com/nobodysu/zabbix-smartmontools

nobodysu, 2019-03-02:
It will work after configuration as described in readme. Can't say more before seeing the outputs:
`zabbix_get -s 192.0.2.1 -k mini.disktemp.discovery[get,"Example host"]`
`zabbix_get -s 192.0.2.1 -k mini.disktemp.discovery[getverb,"Example host"]`

CheckB, 2019-03-02:
```
root@Test:~# zabbix_get -s 192.168.1.4 -k mini.disktemp.discovery[get,"Example host"]
{
    "data": [
        {
            "{#DISK}": "sda"
        },
        {
            "{#DISK}": "sdb"
        },
        {
            "{#DISK}": "sdc"
        }
    ]
}
root@Test:~# zabbix_get -s 192.168.1.4 -k mini.disktemp.discovery[getverb,"Example host"]
  Python version:
 3.5.3 (default, Sep 27 2018, 17:25:39)
[GCC 6.3.0 20170516]

  Sender version:
 zabbix_sender (Zabbix) 4.0.5
Revision 90164 25 February 2019, compilation time: Feb 25 2019 09:25:45

Copyright (C) 2019 Zabbix SIA
License GPLv2+: GNU GPL version 2 or later <http://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it according to
the license. There is NO WARRANTY, to the extent permitted by law.

This product includes software developed by the OpenSSL Project
for use in the OpenSSL Toolkit (http://www.openssl.org/).

Compiled with OpenSSL 1.1.0f  25 May 2017
Running with OpenSSL 1.1.0j  20 Nov 2018

  Config's main settings:
Server=192.168.1.155
ServerActive=192.168.1.155
Could not find 'Timeout' manual setting in config!
Default value is too low for this script.

  Please report any issues or missing features to:
https://github.com/nobodysu/zabbix-mini-IPMI/issues

zabbix_sender [29602]: DEBUG: answer [{"response":"success","info":"processed: 0; failed: 7; total: 7; seconds spent: 0.000190"}]
info from server: "processed: 0; failed: 7; total: 7; seconds spent: 0.000190"
sent: 7; skipped: 0; total: 7

  Note: the sender will fail if server did not gather LLD previously.

  Data sent to zabbix sender:

"Example host" mini.disk.temp[sda] 29
"Example host" mini.disk.info[sda,DriveStatus] "PROCESSED"
"Example host" mini.disk.temp[sdb] 30
"Example host" mini.disk.info[sdb,DriveStatus] "PROCESSED"
"Example host" mini.disk.info[sdc,DriveStatus] "ERR_CODE_2"
"Example host" mini.disk.temp[MAX] 30
"Example host" mini.disk.info[ConfigStatus] "CONFIGURED"
```

CheckB, 2019-03-02:
Zabbix-sender rolled back to 3.0

nobodysu, 2019-03-02:
Seems OK. Unlink and clear, reassign the template and wait for 22-44+2 minutes. It should work.

CheckB, 2019-03-02:
The problem is the same - there is no information on smart. I look in "latest data"

nobodysu, 2019-03-02:
There will be no information on smart it temperature monitoring script. Please provide screenshots.

CheckB, 2019-03-02:
Why then need the script mini_ipmi_smartctl.py?

CheckB, 2019-03-02:
As prerequisites you need python3, lm-sensors, smartmontools
<<
And also indicated in the requirements of smartmontools

nobodysu, 2019-03-02:
`mini_ipmi_smartctl.py` is for disk temperature monitoring.

CheckB, 2019-03-02:
Thank you, I understand everything.

closed 2019-03-02 by CheckB