nobody43/zabbix-mini-IPMI #39: issue with discovery?

killmasta93, 2019-12-13:
Hi,
the CPU temperature does not appear as in the discovery rule it shows this error see picture

[image]

Thank you

nobodysu, 2019-12-13:
I need a full error output e.g. `./mini_ipmi_lmsensors.py get NON` and `sensors -u` output.

nobodysu, 2019-12-13:
You should try new version also: https://github.com/nobodysu/zabbix-mini-IPMI/pull/38

killmasta93, 2019-12-13:
Thanks for the reply, this is what i get 
root@prometheus4:/etc/zabbix/scripts# ./mini_ipmi_lmsensors.py get NON
Traceback (most recent call last):
  File "./mini_ipmi_lmsensors.py", line 272, in <module>
    getOutput_Out = getOutput()
  File "./mini_ipmi_lmsensors.py", line 68, in getOutput
    pSplit = p.strip()
AttributeError: 'NoneType' object has no attribute 'strip'

as for the update i installed it around 4 days ago

Thank you

nobodysu, 2019-12-13:
lmsensors not installed? Fix incoming.

killmasta93, 2019-12-14:
So it seems i needed to install 
sudo apt-get install lm-sensors
after that i ran root@prometheus4:/etc/zabbix/scripts# ./mini_ipmi_lmsensors.py get NON

root@prometheus4:/etc/zabbix/scripts# ./mini_ipmi_lmsensors.py get NON
{
    "data": [
        {
            "{#CPU}": 0
        },
        {
            "{#CORE}": "0",
            "{#CPUC}": 0
        },
        {
            "{#CORE}": "1",
            "{#CPUC}": 0
        },
        {
            "{#CORE}": "2",
            "{#CPUC}": 0
        },
        {
            "{#CORE}": "3",
            "{#CPUC}": 0
        }
    ]
}

I guess i need to wait till it refreshes on zabbix

killmasta93, 2019-12-14:
after a while im seeing the data coming that was it needed to install lm sensors

killmasta93, 2019-12-14:
just a quick question, as for the temperature threshold is it possible to change it? or is it possible to change it for certain hosts?

closed 2019-12-14 by killmasta93

nobodysu, 2019-12-14:
I'm referring to lmsensors in the [latest version](https://github.com/nobodysu/zabbix-mini-IPMI/tree/cpu_refactoring).
Tjmax could not be changed.
In the template there are a `Macros` tab for global thresholds.
In specific host there are `Macros - Inherited and host macros` section for host-specific thresholds.

killmasta93, 2019-12-14:
Thank you for the quick reply, i ended up changing it on the trigger section which says 

  | {Template mini-IPMI v2:mini.disk.temp[MAX].last(500)}>46
-- | --
i changed the value to 52

commit e66ddea6 (2019-12-14): CPU refactoring part 1 (#38)

* dropping zabbix24 support
* switching to special OHMR which could exclude HDDs: openhardwaremonitor/openhardwaremonitor#1115
* disabling non-major triggers to reduce noise
* template tuning via user macro
* max vcore dropped
* max vtt dropped
* complex board and cpu-specific rules dropped
* ryzen support on linux
* fixes #19
* fixes #21
* fixes #29
* fixes #39