nobody43/zabbix-mini-IPMI #41: issue on CPU temperature

killmasta93, 2019-12-22:
Hi,
Recently zabbix started to push this error as its not reading the CPU temperature was working a week ago fine, having lm-sensors

see picture 
Thank you 
[image]

nobodysu, 2019-12-22:
1. Use the [latest version](https://github.com/nobodysu/zabbix-mini-IPMI/blob/master/Linux/mini_ipmi_lmsensors.py)
2. Provide  FULL output of `/etc/zabbix/scripts/mini_ipmi_lmsensors.py getverb NON`

killmasta93, 2019-12-22:
Thanks for the quick reply so after updating on the discovery it disappeard the issue but on the console shows this picture 
[image]

as for the command i got this
```
root@prometheus:~# /etc/zabbix/scripts/mini_ipmi_lmsensors.py getverb NON
  Python version:
 3.5.3 (default, Sep 27 2018, 17:25:39) 
[GCC 6.3.0 20170516]

  Sender version:
 zabbix_sender (Zabbix) 3.0.7
Revision 64609 21 December 2016, compilation time: Jun  4 2017 07:14:06

Copyright (C) 2016 Zabbix SIA
License GPLv2+: GNU GPL version 2 or later <http://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it according to
the license. There is NO WARRANTY, to the extent permitted by law.

  Config's main settings:
Server=monitor.mydomain.com
ServerActive=monitor.mydomain.com
Could not find 'Timeout' manual setting in config!
Default value is too low for this script.

  Please report any issues or missing features to:
https://github.com/nobodysu/zabbix-mini-IPMI/issues

root@prometheus:~# 
  Note: the sender will fail if server did not gather LLD previously.

  Data sent to zabbix sender:

"NON" mini.cpu.info[ConfigStatus] "OS_NOCMD"
zabbix_sender [6371]: DEBUG: answer [{"response":"success","info":"processed: 0; failed: 1; total: 1; seconds spent: 0.000015"}]
info from server: "processed: 0; failed: 1; total: 1; seconds spent: 0.000015"
sent: 1; skipped: 0; total: 1
```

nobodysu, 2019-12-22:
lmsensors not installed

killmasta93, 2019-12-22:
o wow that was awkward i  swear last week was working i guess something happened with zfs snapshots prob went 2 weeks before it was installed lmsensors thank you again just installed it waiting for it to update the new info to the zabbix console

closed 2019-12-22 by killmasta93