nobody43/zabbix-mini-IPMI #42: GPU temperature not showing

jsa91, 2020-01-09:
Getting readings of all the core temperatures as i should into Zabbix. 
Unfortunately the "All GPUs maximum temperature" row is empty. 

lm-sensors gives me:
```
$ sensors
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +38.0°C  (high = +85.0°C, crit = +105.0°C)
Core 0:        +27.0°C  (high = +85.0°C, crit = +105.0°C)
Core 1:        +33.0°C  (high = +85.0°C, crit = +105.0°C)
Core 2:        +32.0°C  (high = +85.0°C, crit = +105.0°C)
Core 3:        +39.0°C  (high = +85.0°C, crit = +105.0°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +27.8°C  (crit = +106.0°C)
temp2:        +29.8°C  (crit = +106.0°C)
temp3:        +30.0°C  (crit = +110.0°C)

radeon-pci-0100
Adapter: PCI adapter
temp1:        +53.5°C  (crit = +120.0°C, hyst = +90.0°C)
```
lm-sensor finds my radeon-pci-0100 but Zabbix does not.
Am i missing something here?

nobodysu, 2020-01-09:
https://github.com/nobodysu/zabbix-mini-IPMI/blob/master/Linux/mini_ipmi_lmsensors.py#L169

`temp = re.search(r'(nouveau.+|nvidia.+|radeon.+)\n.+\n.+\n(?:\s+)?temp\d+_input:\s+(\d+)', i, re.I)`

And what if you change this line?

jsa91, 2020-01-09:
Yep, that worked! Thank you very much for the help.

nobodysu, 2020-01-09:
That's great!
Will reopen until pushed.