nobody43/zabbix-mini-IPMI #43: Many errors in the document README/INSTALL/CONFIG

nicolasgoudard, 2020-02-09:
**Describe the bug**

I don't know how to install  OpenHardwareMonitorReport on Linux as required in your documentations and  I miss data for the voltage and fans speed, but I get good datas for the CPUS: 
in my zabbix interface : "mini-IPMI: Voltage"  and "Fanspeed " are empty
Below my issue, you can see the screenshots

The documentation (Readme)  is not accurate :  missing Zabbix versions requirements and how to fix the paths
My system : 

\# cat /etc/redhat-release
 CentOS Linux release 8.1.1911 (Core) 

zabbix_server --version
zabbix_server (Zabbix) 4.4.5

The second step tell to add an Include path in  zabbix_server.conf  but it is not necessary : The default "Include=/etc/zabbix/zabbix_agentd.d/*.conf" is sufficient 
Moreover, you tell to put the .py files in the folder /etc/zabbix/scripts/ but the default script path is specified in the file   /etc/zabbix/zabbix_server.conf 
"ExternalScripts=/usr/lib/zabbix/externalscripts"
so we must put the files in this folder 

Your documentation to install your project and copy the files is bad explained for a newbie
So  the correct way, for Linux,  is  : 
dnf install -y lm_sensors;
sensors-detect --auto;
dnf install -y smartmontools

STEP Zero
Import Template_mini-IPMI_v2.xml in zabbix web interface : open Zabbix web interface as administrator account : go to menu Configuration / Templates then click on the import button 
Then on Configuration / hosts, attach the template "Template mini-IPMI v2" for every hosts in their template tab
Install prerequites : 
\# dnf -y install zabbix-sender zabbix-get
\# dnf install -y python3
\# dnf install -y lm_sensors;
\# sensors-detect --auto;
\# dnf install -y smartmontools
\# dnf -y install sudo

First step 
\# cd /tmp
\# git clone https://github.com/nobodysu/zabbix-mini-IPMI
\# cd cd /tmp/zabbix-mini-IPMI/
\# cp  mini_ipmi_smartctl.py Linux/mini_ipmi_lmsensors.py sender_wrapper.py
\# cp Linux/sudoers.d/zabbix /etc/sudoers.d/ 
\# cp  Linux/zabbix_agentd.d/userparameter_mini-ipmi2.conf  /etc/zabbix/zabbix_agentd.d/

Second step :
\# cd /usr/lib/zabbix/externalscripts
\# chmod 755 mini_ipmi*.py sender_wrapper.py 
\# chown root:zabbix mini_ipmi*.py sender_wrapper.py 
\# cd /etc/zabbix/zabbix_agentd.d/
\# chmod 644 root:zabbix userparameter_mini-ipmi2.conf
\# chmod 400 /etc/sudoers.d/zabbix
\# chown root /etc/sudoers.d/zabbix

THIRD STEP MISSING : 
check and correct the fix paths in every files to /usr/lib/zabbix/externalscripts/  
fix the path in the configuration file : 
\# vi /etc/zabbix/zabbix_agentd.d/userparameter_mini-ipmi2.conf 
then fixture paths in  the variables declarations  SENDER_WRAPPER_PATH and AGENT_CONF_PATH in 
\#  vi /usr/lib/zabbix/externalscripts/mini_ipmi*.py

Testing 
From the server : 
\# zabbix_get -s IP_AGENT -k mini.cputemp.discovery[get,"HOSTNAME_AGENT"]
\# zabbix_get -s IP_AGENT -k mini.disktemp.discovery[get,"HOSTNAME_AGENT"]
\# zabbix_get -s IP_AGENT -k mini.cputemp.discovery[getverb,"HOSTNAME_AGENT"]
\# zabbix_get -s IP_AGENT -k mini.disktemp.discovery[getverb,"HOSTNAME_AGENT"]

Why I have not datas for the voltages and fan speed ? 

Thanks you in advance

Cordially

[
[image]
[image]
[image]
[image]
[image]

](url)

nobodysu, 2020-02-09:
I'm considering small improvements to the readme, but it can't be extremely verbose - it's not a Linux basics tutorial after all.
About your problem. Please provide outputs:
`sensors -u`
`zabbix_get -s IP_AGENT -k mini.cputemp.discovery[getverb,"HOSTNAME_AGENT"]`

nicolasgoudard, 2020-02-09:
Thanks for the reply and for your time to help me. Here the output : 
[root@alta ~]# zabbix_get -s plana  -k mini.cputemp.discovery[getverb,"plana"]
  Python version:
 3.6.8 (default, Nov 21 2019, 19:31:34) 
[GCC 8.3.1 20190507 (Red Hat 8.3.1-4)]

  Sender version:
 zabbix_sender (Zabbix) 4.4.5
Revision b93f5c4fc0 28 January 2020, compilation time: Jan 29 2020 19:07:36

Copyright (C) 2020 Zabbix SIA
License GPLv2+: GNU GPL version 2 or later <http://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it according to
the license. There is NO WARRANTY, to the extent permitted by law.

This product includes software developed by the OpenSSL Project
for use in the OpenSSL Toolkit (http://www.openssl.org/).

Compiled with OpenSSL 1.1.1 FIPS  11 Sep 2018
Running with OpenSSL 1.1.1c FIPS  28 May 2019

  Config's main settings:
Server=10.193.10.40
ServerActive=10.193.10.40
Could not find 'Timeout' manual setting in config!
Default value is too low for this script.

  Please report any issues or missing features to:
https://github.com/nobodysu/zabbix-mini-IPMI/issues

zabbix_sender [26182]: DEBUG: answer [{"response":"success","info":"processed: 70; failed: 0; total: 70; seconds spent: 0.002968"}]
Response from "10.193.10.40:10051": "processed: 70; failed: 0; total: 70; seconds spent: 0.002968"
sent: 70; skipped: 0; total: 70

  Note: the sender will fail if server did not gather LLD previously.

  Data sent to zabbix sender:

"plana" mini.cpu.info[cpu0,ID] "coretemp-isa-0003"
"plana" mini.cpu.info[cpu0,TjMax] "92"
"plana" mini.cpu.temp[cpu0,core0] "45"
"plana" mini.cpu.temp[cpu0,core1] "41"
"plana" mini.cpu.temp[cpu0,core2] "38"
"plana" mini.cpu.temp[cpu0,core3] "35"
"plana" mini.cpu.temp[cpu0,core4] "36"
"plana" mini.cpu.temp[cpu0,core5] "37"
"plana" mini.cpu.temp[cpu0,core6] "39"
"plana" mini.cpu.temp[cpu0,core8] "38"
"plana" mini.cpu.temp[cpu0,core9] "36"
"plana" mini.cpu.temp[cpu0,core10] "36"
"plana" mini.cpu.temp[cpu0,core11] "38"
"plana" mini.cpu.temp[cpu0,core12] "36"
"plana" mini.cpu.temp[cpu0,core13] "35"
"plana" mini.cpu.temp[cpu0,core14] "36"
"plana" mini.cpu.temp[cpu0,MAX] "45"
"plana" mini.cpu.info[cpu1,ID] "coretemp-isa-0001"
"plana" mini.cpu.info[cpu1,TjMax] "92"
"plana" mini.cpu.temp[cpu1,core0] "43"
"plana" mini.cpu.temp[cpu1,core1] "43"
"plana" mini.cpu.temp[cpu1,core2] "43"
"plana" mini.cpu.temp[cpu1,core3] "41"
"plana" mini.cpu.temp[cpu1,core4] "42"
"plana" mini.cpu.temp[cpu1,core5] "40"
"plana" mini.cpu.temp[cpu1,core6] "42"
"plana" mini.cpu.temp[cpu1,core8] "46"
"plana" mini.cpu.temp[cpu1,core9] "40"
"plana" mini.cpu.temp[cpu1,core10] "45"
"plana" mini.cpu.temp[cpu1,core11] "44"
"plana" mini.cpu.temp[cpu1,core12] "44"
"plana" mini.cpu.temp[cpu1,core13] "43"
"plana" mini.cpu.temp[cpu1,core14] "41"
"plana" mini.cpu.temp[cpu1,MAX] "46"
"plana" mini.cpu.info[cpu2,ID] "coretemp-isa-0002"
"plana" mini.cpu.info[cpu2,TjMax] "92"
"plana" mini.cpu.temp[cpu2,core0] "40"
"plana" mini.cpu.temp[cpu2,core1] "45"
"plana" mini.cpu.temp[cpu2,core2] "40"
"plana" mini.cpu.temp[cpu2,core3] "38"
"plana" mini.cpu.temp[cpu2,core4] "40"
"plana" mini.cpu.temp[cpu2,core5] "39"
"plana" mini.cpu.temp[cpu2,core6] "45"
"plana" mini.cpu.temp[cpu2,core8] "41"
"plana" mini.cpu.temp[cpu2,core9] "44"
"plana" mini.cpu.temp[cpu2,core10] "39"
"plana" mini.cpu.temp[cpu2,core11] "39"
"plana" mini.cpu.temp[cpu2,core12] "43"
"plana" mini.cpu.temp[cpu2,core13] "38"
"plana" mini.cpu.temp[cpu2,core14] "38"
"plana" mini.cpu.temp[cpu2,MAX] "45"
"plana" mini.cpu.info[cpu3,ID] "coretemp-isa-0000"
"plana" mini.cpu.info[cpu3,TjMax] "92"
"plana" mini.cpu.temp[cpu3,core0] "46"
"plana" mini.cpu.temp[cpu3,core1] "47"
"plana" mini.cpu.temp[cpu3,core2] "44"
"plana" mini.cpu.temp[cpu3,core3] "42"
"plana" mini.cpu.temp[cpu3,core4] "44"
"plana" mini.cpu.temp[cpu3,core5] "43"
"plana" mini.cpu.temp[cpu3,core6] "49"
"plana" mini.cpu.temp[cpu3,core8] "48"
"plana" mini.cpu.temp[cpu3,core9] "43"
"plana" mini.cpu.temp[cpu3,core10] "47"
"plana" mini.cpu.temp[cpu3,core11] "47"
"plana" mini.cpu.temp[cpu3,core12] "42"
"plana" mini.cpu.temp[cpu3,core13] "42"
"plana" mini.cpu.temp[cpu3,core14] "43"
"plana" mini.cpu.temp[cpu3,MAX] "49"
"plana" mini.cpu.temp[MAX] "49"
"plana" mini.cpu.info[ConfigStatus] "NOGPUS"

nobodysu, 2020-02-09:
Can't answer without
`sensors -u`

nicolasgoudard, 2020-02-09:
[nicolas@plana ~]$ sensors -u
coretemp-isa-0003
Adapter: ISA adapter
Package id 3:
  temp1_input: 46.000
  temp1_max: 82.000
  temp1_crit: 92.000
  temp1_crit_alarm: 0.000
Core 0:
  temp2_input: 37.000
  temp2_max: 82.000
  temp2_crit: 92.000
  temp2_crit_alarm: 0.000
Core 1:
  temp3_input: 41.000
  temp3_max: 82.000
  temp3_crit: 92.000
  temp3_crit_alarm: 0.000
Core 2:
  temp4_input: 37.000
  temp4_max: 82.000
  temp4_crit: 92.000
  temp4_crit_alarm: 0.000
Core 3:
  temp5_input: 36.000
  temp5_max: 82.000
  temp5_crit: 92.000
  temp5_crit_alarm: 0.000
Core 4:
  temp6_input: 37.000
  temp6_max: 82.000
  temp6_crit: 92.000
  temp6_crit_alarm: 0.000
Core 5:
  temp7_input: 41.000
  temp7_max: 82.000
  temp7_crit: 92.000
  temp7_crit_alarm: 0.000
Core 6:
  temp8_input: 46.000
  temp8_max: 82.000
  temp8_crit: 92.000
  temp8_crit_alarm: 0.000
Core 8:
  temp10_input: 38.000
  temp10_max: 82.000
  temp10_crit: 92.000
  temp10_crit_alarm: 0.000
Core 9:
  temp11_input: 37.000
  temp11_max: 82.000
  temp11_crit: 92.000
  temp11_crit_alarm: 0.000
Core 10:
  temp12_input: 36.000
  temp12_max: 82.000
  temp12_crit: 92.000
  temp12_crit_alarm: 0.000
Core 11:
  temp13_input: 42.000
  temp13_max: 82.000
  temp13_crit: 92.000
  temp13_crit_alarm: 0.000
Core 12:
  temp14_input: 37.000
  temp14_max: 82.000
  temp14_crit: 92.000
  temp14_crit_alarm: 0.000
Core 13:
  temp15_input: 35.000
[... 224 lines cut ...]
Core 10:
  temp12_input: 43.000
  temp12_max: 82.000
  temp12_crit: 92.000
  temp12_crit_alarm: 0.000
Core 11:
  temp13_input: 48.000
  temp13_max: 82.000
  temp13_crit: 92.000
  temp13_crit_alarm: 0.000
Core 12:
  temp14_input: 46.000
  temp14_max: 82.000
  temp14_crit: 92.000
  temp14_crit_alarm: 0.000
Core 13:
  temp15_input: 42.000
  temp15_max: 82.000
  temp15_crit: 92.000
  temp15_crit_alarm: 0.000
Core 14:
  temp16_input: 44.000
  temp16_max: 82.000
  temp16_crit: 92.000
  temp16_crit_alarm: 0.000

nobodysu, 2020-02-09:
The output does not contain voltage information. You can run interactive `sensors-detect` and accept all probes. Other than that not much could be done.

nicolasgoudard, 2020-02-09:
I do it again but I get the same output  :-(

nobodysu, 2020-02-09:
Then lmsensors does not support your board fully (yet?).

nicolasgoudard, 2020-02-09:
I don't know. How can I do ? I am a newbie. If you see my screenshots, I miss many information  
In the readme you tell install OpenHardwareMonitorReport so I have downloaded it but it seems to be for Windows

nobodysu, 2020-02-09:
That's correct, OHMR is for windows.
Not all motherboard sensors are supported by lmsensors. That's it.

nicolasgoudard, 2020-02-09:
Ok. thanks you.  It is Dell Poweredge R940 machine that we use for a compute cluster on Linux with Slurm in my university.
I have installed zabbix for montoring ; in previous Linux Versions we use Ganglia Web put it does not work in Centos, so I am a newbie . Maybe In need install drivers ? 
A last question :  I don't know how to change the frequency of checking temperature.  When I go in "latest data", the last check was done almost 20 minutes ago.  How can change it to check more often ?

nobodysu, 2020-02-09:
Nope, drivers won't help. Just make sure modules suggested by lmsensors are loaded: https://wiki.archlinux.org/index.php/Lm_sensors
Frequency is tuned at: All templates- > Template mini-IPMI v2 -> Discovery rules -> CPU discovery
(it's not advised to tune disk frequency)

nicolasgoudard, 2020-02-09:
Ok. Thanks you. You can close this problem, I will see later. I have another problem so I open a new issue

closed 2020-02-09 by nobodysu