nobody43/zabbix-mini-IPMI #44: Permissions graphs as guest ?

nicolasgoudard, 2020-02-09:
Hello. Sorry, I don't know If it is a bug or not. When I connect as "guest user" to the zabbix interface , nothing is displayed in the temperature graph (no datas). When I connect in "normal" user administrator it is ok, I have datas.
In the guest user group, the guest group has access to everything in read only...
[read](url
<img width="1073" alt="Capture d’écran 2020-02-09 à 17 16 03" [
[image]
](src="https://user-images.githubusercontent.com/6792673/74105640-e853be00-4b5f-11ea-9e5c-44876bcb557c.png">)
) Only. 

How can I repair it ? Maybe I mistake a permission somewhere in the interface ? 

Thanks in advance
Cordially

nobodysu, 2020-02-09:
Administration - User groups - Guest - Permissions

nicolasgoudard, 2020-02-09:
I empty the caches and it works now.

closed 2020-02-09 by nicolasgoudard