nobody43/zabbix-mini-IPMI #45: change trigger temperature treshold

nicolasgoudard, 2020-02-10:
Hello
I want to change the tresholds for each host before having alerts on my zabbix map
The map tell that the temperature is too high( over 60° ) but my sensors on Dell servers specifications tell that the critical threshold is over 92 (different on each node)
sample :
\# sensors
Core 0:        +57.0°C  (high = +82.0°C, crit = +92.0°C)

How can I do ? (I am newbie in zabbix configuration and I don't know where I can change the variable $CPU.HIGH.TEMP on each node)
thx u in advance
Best regards

 - OS: CentOS 8.1
 - Zabbix server version:  4.4.5

nobodysu, 2020-02-10:
Global: All templates/Template mini-IPMI v2 -> Macros
Per-host: host -> Macros -> Inherited

Critical is a throttling temperature and have it's own trigger.
`$CPU.HIGH.TEMP` is there to take action beforehand. (it's not `high = +82.0°C`)

nicolasgoudard, 2020-02-11:
Ok, thanks you, it is working now

closed 2020-02-11 by nobodysu