nobody43/zabbix-mini-IPMI #46: Disk temperature too slow

nicolasgoudard, 2020-02-19:
Mini ipmi tell the temperature is too low for 20 degrees Celsius on my Poweredge R940 for my disk raid (bus_0_megaraid_1) but the specifications tell that 20 degrees is ok.
How can I change the threshold ? I don't find it in the macros. When I look the configuration I don't see about a treshold

[image]
[image]
[image]

nobodysu, 2020-02-20:
Right now it's hardcoded in the script: https://github.com/nobodysu/zabbix-mini-IPMI/blob/master/mini_ipmi_smartctl.py#L33

closed 2020-02-26 by nobodysu

nicolasgoudard, 2020-02-26:
yes thanks you. It is perfectly working