nobody43/zabbix-mini-IPMI #47: zabbix_agentd[5697]: zabbix_agentd [5697]: ERROR: cannot add user parameter "1": not comma-separated

hardwareadictos, 2020-03-22:
**Describe the bug**
Just added all the stuff needed and configured it, i get:

zabbix_agentd[5697]: zabbix_agentd [5697]: ERROR: cannot add user parameter "1": not comma-separated

Once restarting the agent to apply the config.

**To Reproduce**
Enabling UserParaemters on zabbix agent config.

**Expected behavior**
Agent starting the service correctly.

**Screenshots**
[image]

**Provide all outputs described in [Testing](https://github.com/nobodysu/zabbix-mini-IPMI#testing) step**
Serial numbers should be replaced with X_SERIAL_X.

**Please complete the following information:**
 - OS: Debian 10 and Windows 10
 - Zabbix server version: 4.0.18
 - Active or passive check: Passive
 - Using zabbix proxy:  No proxy

**Additional context**
Those are the only UserParameters i'm using, having the problem with zabbix-smartmontools and zabbix-mini-IPMI

nobodysu, 2020-03-22:
```
$ zabbix_agentd -V
$ cat /etc/zabbix/zabbix_agentd.d/*
```
?

nobodysu, 2020-03-22:
Also try agent 3.0 on Windows.

hardwareadictos, 2020-03-22:
root@AtomikPrometeus-DBN01:~# zabbix_agentd -V
zabbix_agentd (daemon) (Zabbix) 4.0.18
Revision 688cd21456 21 February 2020, compilation time: Feb 21 2020 13:47:34

Copyright (C) 2020 Zabbix SIA
License GPLv2+: GNU GPL version 2 or later <http://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it according to
the license. There is NO WARRANTY, to the extent permitted by law.

This product includes software developed by the OpenSSL Project
for use in the OpenSSL Toolkit (http://www.openssl.org/).

Compiled with OpenSSL 1.1.1c  28 May 2019
Running with OpenSSL 1.1.1d  10 Sep 2019
root@AtomikPrometeus-DBN01:~# cat /etc/zabbix/zabbix_agentd.d/*
UserParameter=mini.disktemp.discovery[*], sudo "/etc/zabbix/scripts/mini_ipmi_smartctl.py" "$1" "$2"
UserParameter=mini.cputemp.discovery[*], "/etc/zabbix/scripts/mini_ipmi_lmsensors.py" "$1" "$2"
UserParameter=smartctl.discovery[*], sudo "/etc/zabbix/scripts/smartctl-lld.py" "$1" "$2"
root@AtomikPrometeus-DBN01:~# ll /etc/zabbix/zabbix_agentd.d/

nobodysu, 2020-03-22:
I can't tell what's wrong here. I've rolled clean install on the same environment and everything works.
Try to install from scratch, including purging the config files.

nobodysu, 2020-04-04:
Hope that's solved.

closed 2020-04-04 by nobodysu