nobody43/zabbix-mini-IPMI #48: GPU temperature on Windows

pedjateg, 2020-04-24:
**Describe the problem**
I cannot get GPU temperatures in zabbix.
Only when I run script manually on client side, there is mini.gpu.temp[MAX] in zabbix.
I tried both active and passive checks and OHM 0.8.0.5 and 0.9.2.0

**To Reproduce**
Activate mini-IPMI Template

**Expected behavior**
To get GPU temperatures on zabbix server

**Screenshots**

**Provide all outputs described in [Testing](https://github.com/nobodysu/zabbix-mini-IPMI#testing) step**

If I run(on server): `zabbix_get -s 10.0.101.101 -k mini.cputemp.discovery[getverb,"HP-PST01"]`

Output:
```
   Python version:
 3.8.1 (tags/v3.8.1:1b293b6, Dec 18 2019, 23:11:46) [MSC v.1916 64 bit (AMD64)]

 Sender version:
 zabbix_sender (Zabbix) 4.2.7
Revision 624fb7497b 2 October 2019, compilation time: Oct  3 2019 14:58:35

  Config's main settings:
Server=192.168.101.21
ServerActive=192.168.101.21
Timeout=10

  Please report any issues or missing features to:
https://github.com/nobodysu/zabbix-mini-IPMI/issues

zabbix_sender.exe [12616]: DEBUG: answer [{"response":"success","info":"processed: 20; failed: 0; total: 20; seconds spent: 0.000570"}]
Response from "192.168.101.21:10051": "processed: 20; failed: 0; total: 20; seconds spent: 0.000570"
sent: 20; skipped: 0; total: 20

  Note: the sender will fail if server did not gather LLD previously.

  Data sent to zabbix sender:

"HP-PST01" mini.info[OHMRversion] "0.8.0.5"
"HP-PST01" mini.brd.info[SMBIOSversion] "2.7"
"HP-PST01" mini.brd.info[BIOSvendor] "Hewlett-Packard"
"HP-PST01" mini.brd.info[BIOSversion] "K01 v02.90"
"HP-PST01" mini.brd.info[MainboardManufacturer] "Hewlett-Packard"
"HP-PST01" mini.brd.info[MainboardName] "3397"
"HP-PST01" mini.brd.info[MainboardVersion] "Processor Manufacturer: Intel"
"HP-PST01" mini.cpu.info[cpu0,ID] "Intel Core i5-3570"
"HP-PST01" mini.cpu.info[cpu0,TjMax] "105"
"HP-PST01" mini.cpu.info[cpu0,CPUstatus] "PROCESSED"
"HP-PST01" mini.cpu.temp[cpu0,core0] "51"
"HP-PST01" mini.cpu.temp[cpu0,core1] "44"
"HP-PST01" mini.cpu.temp[cpu0,core2] "44"
"HP-PST01" mini.cpu.temp[cpu0,core3] "42"
"HP-PST01" mini.cpu.temp[cpu0,MAX] "51"
"HP-PST01" mini.cpu.temp[MAX] "51"
"HP-PST01" mini.gpu.info[gpu0,ID] "AMD Radeon HD 5450"
"HP-PST01" mini.gpu.info[gpu0,GPUstatus] "NO_TEMP"
"HP-PST01" mini.gpu.info[gpu0,GPUstatus] "NO_FAN"
"HP-PST01" mini.cpu.info[ConfigStatus] "NOGPUTEMPS"
```

If I run (on client): `mini_ipmi_ohmr.py getverb HP-PST01`

Output:
```
C:\Program Files\Zabbix Agent\scripts>mini_ipmi_ohmr.py getverb HP-PST01
  Python version:
 3.8.1 (tags/v3.8.1:1b293b6, Dec 18 2019, 23:11:46) [MSC v.1916 64 bit (AMD64)]

  Sender version:
 zabbix_sender (Zabbix) 4.2.7
Revision 624fb7497b 2 October 2019, compilation time: Oct  3 2019 14:58:35

Compiled with OpenSSL 1.1.1  11 Sep 2018
Running with OpenSSL 1.1.1  11 Sep 2018

  Config's main settings:
Server=192.168.101.21
ServerActive=192.168.101.21
Timeout=10

  Please report any issues or missing features to:
https://github.com/nobodysu/zabbix-mini-IPMI/issues

C:\Program Files\Zabbix Agent\scripts>
  Note: the sender will fail if server did not gather LLD previously.

  Data sent to zabbix sender:

"HP-PST01" mini.info[OHMRversion] "0.8.0.5"
"HP-PST01" mini.brd.info[SMBIOSversion] "2.7"
"HP-PST01" mini.brd.info[BIOSvendor] "Hewlett-Packard"
"HP-PST01" mini.brd.info[BIOSversion] "K01 v02.90"
"HP-PST01" mini.brd.info[MainboardManufacturer] "Hewlett-Packard"
"HP-PST01" mini.brd.info[MainboardName] "3397"
"HP-PST01" mini.brd.info[MainboardVersion] "Processor Manufacturer: Intel"
"HP-PST01" mini.cpu.info[cpu0,ID] "Intel Core i5-3570"
"HP-PST01" mini.cpu.info[cpu0,TjMax] "105"
"HP-PST01" mini.cpu.info[cpu0,CPUstatus] "PROCESSED"
"HP-PST01" mini.cpu.temp[cpu0,core0] "50"
"HP-PST01" mini.cpu.temp[cpu0,core1] "42"
"HP-PST01" mini.cpu.temp[cpu0,core2] "44"
"HP-PST01" mini.cpu.temp[cpu0,core3] "46"
"HP-PST01" mini.cpu.temp[cpu0,MAX] "50"
"HP-PST01" mini.cpu.temp[MAX] "50"
"HP-PST01" mini.gpu.info[gpu0,ID] "AMD Radeon HD 5450"
"HP-PST01" mini.gpu.temp[gpu0] "42"
"HP-PST01" mini.gpu.info[gpu0,GPUstatus] "NO_FAN"
"HP-PST01" mini.gpu.temp[MAX] "42"
"HP-PST01" mini.cpu.info[ConfigStatus] "CONFIGURED"
zabbix_sender.exe [14492]: DEBUG: answer [{"response":"success","info":"processed: 20; failed: 1; total: 21; seconds spent: 0.000568"}]
Response from "192.168.101.21:10051": "processed: 20; failed: 1; total: 21; seconds spent: 0.000568"
sent: 21; skipped: 0; total: 21
```

**Please complete the following information:**
 - OS: FreeBSD 11.3-RELEASE-p6
 - Zabbix server version: 4.2.6
 - Active or passive check: both
 - Using zabbix proxy: No proxy
 - UserParameter in zabbix_agentd.conf: 
`UserParameter=mini.disktemp.discovery[*], python "C:\Program Files\Zabbix Agent\scripts\mini_ipmi_smartctl.py" "$1" "$2"`

**Additional context**
Output from OpenHardwareMonitorReport.exe when run scheduled , from zabbix:
```
+- HP 3397 (/mainboard)
|
+- Intel Core i5-3570 (/intelcpu/0)
|  +- Bus Speed      :  99.7733  99.7733  99.7819 (/intelcpu/0/clock/0)
|  +- CPU Core #1    :  1596.37  1596.37  3691.93 (/intelcpu/0/clock/1)
|  +- CPU Core #2    :  1596.37  1596.37  3691.93 (/intelcpu/0/clock/2)
|  +- CPU Core #3    :  1596.37  1596.37  1596.51 (/intelcpu/0/clock/3)
|  +- CPU Core #4    :  1596.37  1596.37  3592.15 (/intelcpu/0/clock/4)
|  +- CPU Core #1    :       50       50       52 (/intelcpu/0/temperature/0)
|  +- CPU Core #2    :       43       41       43 (/intelcpu/0/temperature/1)
|  +- CPU Core #3    :       47       43       47 (/intelcpu/0/temperature/2)
|  +- CPU Core #4    :       43       41       43 (/intelcpu/0/temperature/3)
|  +- CPU Package    :       50       50       52 (/intelcpu/0/temperature/4)
|  +- CPU Total      :  56.2046  30.9028  56.2046 (/intelcpu/0/load/0)
|  +- CPU Core #1    :  57.6923  57.6923      100 (/intelcpu/0/load/1)
|  +- CPU Core #2    :  52.8302     12.5  52.8302 (/intelcpu/0/load/2)
|  +- CPU Core #3    :  57.6923  11.1111  57.6923 (/intelcpu/0/load/3)
|  +- CPU Core #4    :  56.6038        0  56.6038 (/intelcpu/0/load/4)
|  +- CPU Package    :  26.2245  13.4854  26.2245 (/intelcpu/0/power/0)
|  +- CPU Cores      :   22.284  10.0625   22.284 (/intelcpu/0/power/1)
|  +- CPU Graphics   :        0        0        0 (/intelcpu/0/power/2)
|
+- Generic Memory (/ram)
|  +- Memory         :  54.2692  54.2692  54.2692 (/ram/load/0)
|  +- Used Memory    :  4.31212  4.31212  4.31212 (/ram/data/0)
|  +- Available Memory :  3.63367  3.63367  3.63367 (/ram/data/1)
|
+- AMD Radeon HD 5450 (/atigpu/0)
|
+- ST500DM002-1BD142 (/hdd/0)
```
There is no GPU Core (/atigpu/0/temperature/0)

Output from OpenHardwareMonitorReport.exe when I run manually, on client:
```
|
+- HP 3397 (/mainboard)
|
+- Intel Core i5-3570 (/intelcpu/0)
|  +- Bus Speed      :  99.7733  99.7733  99.7785 (/intelcpu/0/clock/0)
|  +- CPU Core #1    :  3591.84  3591.84  3592.03 (/intelcpu/0/clock/1)
|  +- CPU Core #2    :  3591.84  1596.46  3591.84 (/intelcpu/0/clock/2)
|  +- CPU Core #3    :  3591.84  1596.46  3591.84 (/intelcpu/0/clock/3)
|  +- CPU Core #4    :  3691.61  3691.61   3691.8 (/intelcpu/0/clock/4)
|  +- CPU Core #1    :       49       49       51 (/intelcpu/0/temperature/0)
|  +- CPU Core #2    :       42       39       42 (/intelcpu/0/temperature/1)
|  +- CPU Core #3    :       46       39       46 (/intelcpu/0/temperature/2)
|  +- CPU Core #4    :       47       40       47 (/intelcpu/0/temperature/3)
|  +- CPU Package    :       47       47       51 (/intelcpu/0/temperature/4)
|  +- CPU Total      :  44.1177    31.25  44.1177 (/intelcpu/0/load/0)
|  +- CPU Core #1    :   54.902   54.902      100 (/intelcpu/0/load/1)
|  +- CPU Core #2    :  37.2549     12.5  37.2549 (/intelcpu/0/load/2)
|  +- CPU Core #3    :  29.4118     12.5  29.4118 (/intelcpu/0/load/3)
|  +- CPU Core #4    :   54.902        0   54.902 (/intelcpu/0/load/4)
|  +- CPU Package    :  22.7167  15.5279  22.7167 (/intelcpu/0/power/0)
|  +- CPU Cores      :  18.9398  12.0533  18.9398 (/intelcpu/0/power/1)
|  +- CPU Graphics   :        0        0        0 (/intelcpu/0/power/2)
|
+- Generic Memory (/ram)
|  +- Memory         :  52.5117  52.5117  52.5117 (/ram/load/0)
|  +- Used Memory    :  4.17246  4.17246  4.17246 (/ram/data/0)
|  +- Available Memory :  3.77332  3.77332  3.77332 (/ram/data/1)
|
+- AMD Radeon HD 5450 (/atigpu/0)
|  +- GPU Core       :      0.9      0.9      0.9 (/atigpu/0/voltage/0)
|  +- GPU Core       :      157      157      157 (/atigpu/0/clock/0)
|  +- GPU Memory     :      200      200      200 (/atigpu/0/clock/1)
|  +- GPU Core       :       42       42       42 (/atigpu/0/temperature/0)
|  +- GPU Core       :        0        0        0 (/atigpu/0/load/0)
|  +- GPU Fan        :      100      100      100 (/atigpu/0/control/0)
|
+- ST500DM002-1BD142 (/hdd/0)
```

There is GPU Core (/atigpu/0/temperature/0)

nobodysu, 2020-04-24:
Please show me PARAMS:
`PARAMS = 'reporttoconsole --IgnoreMonitorHDD --IgnoreMonitorRAM'`

pedjateg, 2020-04-24:
`PARAMS = 'reporttoconsole --IgnoreMonitorHDD --IgnoreMonitorRAM'`

nobodysu, 2020-04-24:
Did you run OHMR with those params to get the output?
Try to remove `--Ignore` ones.

pedjateg, 2020-04-24:
Yes,  i tried.
Same:
```
"HP-PST01" mini.gpu.info[gpu0,ID] "AMD Radeon HD 5450"
"HP-PST01" mini.gpu.info[gpu0,GPUstatus] "NO_TEMP"
"HP-PST01" mini.gpu.info[gpu0,GPUstatus] "NO_FAN"
"HP-PST01" mini.cpu.info[ConfigStatus] "NOGPUTEMPS"
```

nobodysu, 2020-04-24:
Are you using one of these binaries?
https://github.com/openhardwaremonitor/openhardwaremonitor/pull/1115#issuecomment-462141642
https://github.com/openhardwaremonitor/openhardwaremonitor/pull/1115#issuecomment-616230088

pedjateg, 2020-04-24:
Yes, both.

nobodysu, 2020-04-24:
Why there's /ram and /hdd in outputs then? Could you recheck?

pedjateg, 2020-04-24:
It could be when I set:
```PARAMS = 'reporttoconsole' ```

To understand what happens , I inserted :   
```    pRunStatus = p_Output[0]
    pOut = p_Output[1]
    with open("C:\\zabbix-agent\\scripts\\out.txt", "w") as text_file:
        print("Out: {}".format(pOut), file=text_file) 
```
and when I run 
```zabbix_get -s 10.0.101.101 -k `mini.cputemp.discovery[getverb,"HP-PST01"]```
output in out.txt is:
```
--------------------------------------------------------------------------------
Sensors

|
+- HP 3397 (/mainboard)
|
+- Intel Core i5-3570 (/intelcpu/0)
|  +- Bus Speed      :  99.7763  99.7763  99.7763 (/intelcpu/0/clock/0)
|  +- CPU Core #1    :  3691.72  1596.42  3691.72 (/intelcpu/0/clock/1)
|  +- CPU Core #2    :   3791.5  1596.42   3791.5 (/intelcpu/0/clock/2)
|  +- CPU Core #3    :  1596.42  1596.42  3691.72 (/intelcpu/0/clock/3)
|  +- CPU Core #4    :  1596.42  1596.42  1596.42 (/intelcpu/0/clock/4)
|  +- CPU Core #1    :       49       49       50 (/intelcpu/0/temperature/0)
|  +- CPU Core #2    :       38       38       38 (/intelcpu/0/temperature/1)
|  +- CPU Core #3    :       42       39       42 (/intelcpu/0/temperature/2)
|  +- CPU Core #4    :       37       37       39 (/intelcpu/0/temperature/3)
|  +- CPU Package    :       49       49       50 (/intelcpu/0/temperature/4)
|  +- CPU Total      :  16.6667  16.6667       25 (/intelcpu/0/load/0)
|  +- CPU Core #1    :  33.3333  33.3333      100 (/intelcpu/0/load/1)
|  +- CPU Core #2    :  33.3333        0  33.3333 (/intelcpu/0/load/2)
|  +- CPU Core #3    :        0        0        0 (/intelcpu/0/load/3)
|  +- CPU Core #4    :        0        0        0 (/intelcpu/0/load/4)
|  +- CPU Package    :  18.6091  13.4898  18.6091 (/intelcpu/0/power/0)
|  +- CPU Cores      :  14.9143  9.85262  14.9143 (/intelcpu/0/power/1)
|  +- CPU Graphics   :        0        0        0 (/intelcpu/0/power/2)
|
+- AMD Radeon HD 5450 (/atigpu/0)

--------------------------------------------------------------------------------

Parameters

|
+- HP 3397 (/mainboard)
|
+- Intel Core i5-3570 (/intelcpu/0)
|  |
|  +- CPU Core #1 (/intelcpu/0/temperature/0)
|  |  +- TjMax [øC] : 105 : 105
|  |  +- TSlope [øC] : 1 : 1
|  |
|  +- CPU Core #2 (/intelcpu/0/temperature/1)
|  |  +- TjMax [øC] : 105 : 105
|  |  +- TSlope [øC] : 1 : 1
|  |
|  +- CPU Core #3 (/intelcpu/0/temperature/2)
|  |  +- TjMax [øC] : 105 : 105
|  |  +- TSlope [øC] : 1 : 1
|  |
|  +- CPU Core #4 (/intelcpu/0/temperature/3)
|  |  +- TjMax [øC] : 105 : 105
|  |  +- TSlope [øC] : 1 : 1
|  |
|  +- CPU Package (/intelcpu/0/temperature/4)
|  |  +- TjMax [øC] : 105 : 105
|  |  +- TSlope [øC] : 1 : 1
|
+- AMD Radeon HD 5450 (/atigpu/0)

--------------------------------------------------------------------------------
```

nobodysu, 2020-04-24:
Does zabbix agent service runs from 'Local System account'? Has the account been altered or administrative permissions been revoked?

pedjateg, 2020-04-24:
Yes, zabbix is running as service, as Local system Account.
My account is only account on system, and have administrative privileges.
I tried run zabbix service as This account, but result is same.

pedjateg, 2020-04-24:
I inserted     
```
    os.chdir('C:\\Tools\\ohm')
    os.system('"OpenHardwareMonitorReport.exe" ReportToconsole')
   ``` 
and when I run from server, 
```zabbix_get -s 10.0.101.101 -k mini.cputemp.discovery[getverb,"HP-PST01"]```
in console I get:
```
--------------------------------------------------------------------------------

Sensors

|
+- HP 3397 (/mainboard)
|
+- Intel Core i5-3570 (/intelcpu/0)
|  +- Bus Speed      :  99.7733  99.7733  99.7784 (/intelcpu/0/clock/0)
|  +- CPU Core #1    :  3591.84  3591.84   3691.8 (/intelcpu/0/clock/1)
|  +- CPU Core #2    :  3591.84  3591.84  3592.02 (/intelcpu/0/clock/2)
|  +- CPU Core #3    :  3591.84  3591.84   3691.8 (/intelcpu/0/clock/3)
|  +- CPU Core #4    :  3591.84  1596.45  3591.84 (/intelcpu/0/clock/4)
|  +- CPU Core #1    :       50       50       51 (/intelcpu/0/temperature/0)
|  +- CPU Core #2    :       46       39       46 (/intelcpu/0/temperature/1)
|  +- CPU Core #3    :       48       40       48 (/intelcpu/0/temperature/2)
|  +- CPU Core #4    :       41       41       41 (/intelcpu/0/temperature/3)
|  +- CPU Package    :       50       50       51 (/intelcpu/0/temperature/4)
|  +- CPU Total      :  53.9216       25  53.9216 (/intelcpu/0/load/0)
|  +- CPU Core #1    :   45.098   45.098       50 (/intelcpu/0/load/1)
|  +- CPU Core #2    :   54.902        0   54.902 (/intelcpu/0/load/2)
|  +- CPU Core #3    :  52.9412        0  52.9412 (/intelcpu/0/load/3)
|  +- CPU Core #4    :  62.7451       50  62.7451 (/intelcpu/0/load/4)
|  +- CPU Package    :  25.9713  12.8268  25.9713 (/intelcpu/0/power/0)
|  +- CPU Cores      :  22.1557  9.48745  22.1557 (/intelcpu/0/power/1)
|  +- CPU Graphics   :        0        0        0 (/intelcpu/0/power/2)
|
+- Generic Memory (/ram)
|  +- Memory         :  61.2709  61.2709  61.2709 (/ram/load/0)
|  +- Used Memory    :  4.86845  4.86845  4.86845 (/ram/data/0)
|  +- Available Memory :  3.07733  3.07733  3.07733 (/ram/data/1)
|
+- AMD Radeon HD 5450 (/atigpu/0)
|
+- ST500DM002-1BD142 (/hdd/0)
|  +- Temperature    :       36       36       36 (/hdd/0/temperature/0)
|  +- Used Space     :  55.7897  55.7897  55.7897 (/hdd/0/load/0)
|
+- KINGSTON SVP200S37A120G (/hdd/1)
|  +- Temperature    :       30       30       30 (/hdd/1/temperature/0)
|  +- Used Space     :  91.7151  91.7151  91.7151 (/hdd/1/load/0)
|  +- Remaining Life :       86       86       86 (/hdd/1/level/0)
|  +- Write Amplification :  1.29233  1.29233  1.29233 (/hdd/1/factor/1)
|  +- Controller Writes to NAND :    57241    57241    57241 (/hdd/1/data/0)
|  +- Host Writes to Controller :    44293    44293    44293 (/hdd/1/data/1)
|  +- Host Reads     :    37494    37494    37494 (/hdd/1/data/2)
```

When I run from client,
 ```mini_ipmi_ohmr.py getverb HP-PST01``` 
in console I get :
```
--------------------------------------------------------------------------------

Sensors

|
+- HP 3397 (/mainboard)
|
+- Intel Core i5-3570 (/intelcpu/0)
|  +- Bus Speed      :  99.7733  99.7733  99.7773 (/intelcpu/0/clock/0)
|  +- CPU Core #1    :  3591.84  3591.84  3591.98 (/intelcpu/0/clock/1)
|  +- CPU Core #2    :  3791.39  1596.44  3791.39 (/intelcpu/0/clock/2)
|  +- CPU Core #3    :  3591.84  1596.44  3591.84 (/intelcpu/0/clock/3)
|  +- CPU Core #4    :  3591.84  3591.84  3691.76 (/intelcpu/0/clock/4)
|  +- CPU Core #1    :       50       50       51 (/intelcpu/0/temperature/0)
|  +- CPU Core #2    :       40       40       40 (/intelcpu/0/temperature/1)
|  +- CPU Core #3    :       46       41       46 (/intelcpu/0/temperature/2)
|  +- CPU Core #4    :       41       41       41 (/intelcpu/0/temperature/3)
|  +- CPU Package    :       50       50       51 (/intelcpu/0/temperature/4)
|  +- CPU Total      :  38.7755     37.5  38.7755 (/intelcpu/0/load/0)
|  +- CPU Core #1    :  36.7347  36.7347       75 (/intelcpu/0/load/1)
|  +- CPU Core #2    :  32.6531       25  32.6531 (/intelcpu/0/load/2)
|  +- CPU Core #3    :  48.9796        0  48.9796 (/intelcpu/0/load/3)
|  +- CPU Core #4    :  36.7347  36.7347       50 (/intelcpu/0/load/4)
|  +- CPU Package    :  22.9581  17.0407  22.9581 (/intelcpu/0/power/0)
|  +- CPU Cores      :  19.1994   11.162  19.1994 (/intelcpu/0/power/1)
|  +- CPU Graphics   :        0        0        0 (/intelcpu/0/power/2)
|
+- Generic Memory (/ram)
|  +- Memory         :  61.7843  61.7843  61.7843 (/ram/load/0)
|  +- Used Memory    :  4.90925  4.90925  4.90925 (/ram/data/0)
|  +- Available Memory :  3.03654  3.03654  3.03654 (/ram/data/1)
|
+- AMD Radeon HD 5450 (/atigpu/0)
|  +- GPU Core       :        1        1        1 (/atigpu/0/voltage/0)
|  +- GPU Core       :      650      650      650 (/atigpu/0/clock/0)
|  +- GPU Memory     :      400      400      400 (/atigpu/0/clock/1)
|  +- GPU Core       :     44.5     44.5     44.5 (/atigpu/0/temperature/0)
|  +- GPU Core       :        0        0        0 (/atigpu/0/load/0)
|  +- GPU Fan        :      100      100      100 (/atigpu/0/control/0)
|
+- ST500DM002-1BD142 (/hdd/0)
|  +- Temperature    :       36       36       36 (/hdd/0/temperature/0)
|  +- Used Space     :  55.7897  55.7897  55.7897 (/hdd/0/load/0)
|
+- KINGSTON SVP200S37A120G (/hdd/1)
|  +- Temperature    :       30       30       30 (/hdd/1/temperature/0)
|  +- Used Space     :  91.7155  91.7155  91.7155 (/hdd/1/load/0)
|  +- Remaining Life :       86       86       86 (/hdd/1/level/0)
|  +- Write Amplification :  1.29233  1.29233  1.29233 (/hdd/1/factor/1)
|  +- Controller Writes to NAND :    57241    57241    57241 (/hdd/1/data/0)
|  +- Host Writes to Controller :    44293    44293    44293 (/hdd/1/data/1)
|  +- Host Reads     :    37494    37494    37494 (/hdd/1/data/2)

--------------------------------------------------------------------------------
```

pedjateg, 2020-04-24:
This could be problem ?
https://github.com/openhardwaremonitor/openhardwaremonitor/issues/33#issuecomment-133939708

nobodysu, 2020-04-24:
At this point I'm afraid I can't help with this problem. It's not related to the script, and I'm not familiar with OHMR codebase/C#.

> Could this be a problem ?

It's possible, but I have NVIDIA card on Windows which reporting temperature properly.
Have you tried different driver?

pedjateg, 2020-04-24:
Simailar problem is when I am run OHM as a service, there is no GPU sensors.

- activate remote Web server on OHM (Options - remote webserver - run) and close OHM
- download NSSM - https://nssm.cc/download
- ```nssm install ohm "C:\Tools\ohm\OpenHardwareMonitor.exe"```
- ```nssm start ohm```
- open http://127.0.0.1:8085/

pedjateg, 2020-04-24:
>  Have you tried different driver?

I tried with new driver, same problem.
It looks like problem is with OpenHardwareMonitor and starting by service in Session 0.

Thank you very much for help :)

mflanagan93, 2020-10-29:
I know this is a little old but has there been any solution to this problem? I am new to Zabbix and am now experiencing this same issue. I can get the GPU Temp if i run the script from the host but not if I query from the server. Thanks in advance.

nobodysu, 2020-10-29:
Nope, the solution must come from Windows or OHMR itself.
However, you could assign the regular run in Scheduled Tasks.

mflanagan93, 2020-10-29:
its very perplexing when ran on the client its fine when requested from the server its not. Agreed that would be a workaround. Thanks for the quick reply.

mflanagan93, 2020-11-02:
Apparently to get the GPU Temp the request must come from a logged in user. I tried to use the task scheduler as a solution but was faced with the same exact problem as before as I was selecting run task even though not logged in because the user accounts don't have privileges (by design) to execute tasks. So the fault is with OHM.