nobody43/zabbix-mini-IPMI #49: "Unexpected BIOS firmware version" Trigger meaning

cyberjacob, 2020-05-01:
**Describe the problem**
Can anyone give more information on what the "Unexpected BIOS firmware version" trigger is reporting? I've had this alert show up for a Windows PC today, but can't find what problem it's actually trying to tell me about.

**To Reproduce**
Unknown

**Expected behavior**
Unknown

**Screenshots**
[image]

**Please complete the following information:**
 - OS: Windows 10
 - Zabbix server version: 4.4
 - Active or passive check: Passive
 - Using zabbix proxy: No

nobodysu, 2020-05-02:
It's intended to place the list of BIOS versions on your network in `{$EXPECTED.BIOS.VERS}` macro inside template. This way you'll know (by maintaining the list of newest ones) when machines are outdated and needing a BIOS update.

nobodysu, 2020-06-08:
Hope that's resolved.

closed 2020-06-08 by nobodysu