nobody43/zabbix-mini-IPMI #50: Tempeture not showing on HP server ML350GEN9

killmasta93, 2020-05-30:
**Describe the bug**
Does not show disk temp

**To Reproduce**
installed lm sensors but it does show CPU temperature 

**Screenshots**
[image]

**Provide all outputs described in [Testing](https://github.com/nobodysu/zabbix-mini-IPMI#testing) step**
Serial numbers should be replaced with X_SERIAL_X.

**Please complete the following information:**
 - OS: [e.g. Debian] debian 10
 - Zabbix server version: [e.g. 3.0] 4.4
 - Active or passive check: [e.g. Passive] active
 - Using zabbix proxy: [e.g. No proxy] none

nobodysu, 2020-05-30:
`zabbix_get -s 192.0.2.1 -k mini.cputemp.discovery[getverb,"Example host"]`
`zabbix_get -s 192.0.2.1 -k mini.disktemp.discovery[getverb,"Example host"]`

killmasta93, 2020-05-30:
thanks for the reply, quick question i should run this from the server of zabbix or on the zabbix agent? Some odd reason im getting 

`zabbix_get [14422]: Timeout while executing operation`

nobodysu, 2020-05-30:
Server or proxy, if any.
>zabbix_get [14422]: Timeout while executing operation

Increase `Timeout`.
>mini.cputemp.discovery

you need `disktemp`

killmasta93, 2020-05-30:
thanks for the reply,
this is what i get

```
root@monitor:~# zabbix_get -s theip -k mini.cputemp.discovery[get,"prometheusaa"]
{
    "data": [
        {
            "{#CPU}": 0
        },
        {
            "{#CPUC}": 0,
            "{#CORE}": "0"
        },
        {
            "{#CPUC}": 0,
            "{#CORE}": "1"
        },
        {
            "{#CPUC}": 0,
            "{#CORE}": "2"
        },
        {
            "{#CPUC}": 0,
            "{#CORE}": "3"
        },
        {
            "{#CPUC}": 0,
            "{#CORE}": "4"
        },
        {
            "{#CPUC}": 0,
            "{#CORE}": "5"
        }
    ]
}
root@monitor:~# zabbix_get -s 190.0.0.186 -k mini.disktemp.discovery[get,"prometheusaa"]
{
    "data": [
        {
            "{#DISK}": "sda"
        },
        {
            "{#DISK}": "sdb"
        },
        {
            "{#DISK}": "sdc"
        },
        {
            "{#DISK}": "sdd"
        },
        {
            "{#DISK}": "sde"
        },
        {
            "{#DISK}": "sdf"
        },
        {
            "{#DISK}": "sdg"
        }
    ]
}

```

nobodysu, 2020-05-31:
getverb?

killmasta93, 2020-05-31:
thanks for the reply this is what i get

```
root@monitor:~# zabbix_get -s 190.0.0.186 -k mini.disktemp.discovery[getverb,"prometheusaa"]
  Path guess: linux

  Python version:
 3.7.3 (default, Dec 20 2019, 18:57:59) 
[GCC 8.3.0]

  Sender version:
 zabbix_sender (Zabbix) 4.0.4
Revision 89349 4 February 2019, compilation time: Feb  6 2019 00:50:40

Copyright (C) 2019 Zabbix SIA
License GPLv2+: GNU GPL version 2 or later <http://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it according to
the license. There is NO WARRANTY, to the extent permitted by law.

Compiled with GnuTLS 3.6.6
Running with GnuTLS 3.6.7

  Config's main settings:
Server=monitor.mydomain.com
ServerActive=monitor.mydomain.com
Timeout=30

  Please report any issues or missing features to:
https://github.com/nobodysu/zabbix-mini-IPMI/issues

zabbix_sender [21455]: DEBUG: answer [{"response":"success","info":"processed: 31; failed: 0; total: 31; seconds spent: 0.002804"}]
info from server: "processed: 31; failed: 0; total: 31; seconds spent: 0.002804"
sent: 31; skipped: 0; total: 31

  Note: the sender will fail if server did not gather LLD previously.

  Data sent to zabbix sender:

"prometheusaa" mini.disk.info[sda,DriveStatus] "NOTEMP"
"prometheusaa" mini.disk.tempMin[sda] "25"
"prometheusaa" mini.disk.tempMax[sda] "45"
"prometheusaa" mini.disk.tempCrit[sda] "60"
"prometheusaa" mini.disk.info[sdb,DriveStatus] "NOTEMP"
"prometheusaa" mini.disk.tempMin[sdb] "25"
"prometheusaa" mini.disk.tempMax[sdb] "45"
"prometheusaa" mini.disk.tempCrit[sdb] "60"
"prometheusaa" mini.disk.info[sdc,DriveStatus] "NOTEMP"
"prometheusaa" mini.disk.tempMin[sdc] "25"
"prometheusaa" mini.disk.tempMax[sdc] "45"
"prometheusaa" mini.disk.tempCrit[sdc] "60"
"prometheusaa" mini.disk.info[sdd,DriveStatus] "NOTEMP"
"prometheusaa" mini.disk.tempMin[sdd] "25"
"prometheusaa" mini.disk.tempMax[sdd] "45"
"prometheusaa" mini.disk.tempCrit[sdd] "60"
"prometheusaa" mini.disk.info[sde,DriveStatus] "NOTEMP"
"prometheusaa" mini.disk.tempMin[sde] "25"
"prometheusaa" mini.disk.tempMax[sde] "45"
"prometheusaa" mini.disk.tempCrit[sde] "60"
"prometheusaa" mini.disk.info[sdf,DriveStatus] "NOTEMP"
"prometheusaa" mini.disk.tempMin[sdf] "25"
"prometheusaa" mini.disk.tempMax[sdf] "45"
"prometheusaa" mini.disk.tempCrit[sdf] "60"
"prometheusaa" mini.disk.info[sdg,DriveStatus] "PROCESSED"
"prometheusaa" mini.disk.temp[sdg] "29"
"prometheusaa" mini.disk.tempMin[sdg] "25"
"prometheusaa" mini.disk.tempMax[sdg] "45"
"prometheusaa" mini.disk.tempCrit[sdg] "60"
"prometheusaa" mini.disk.info[ConfigStatus] "CONFIGURED"
"prometheusaa" mini.disk.temp[MAX] "29"

```

nobodysu, 2020-05-31:
`# smartctl -A -i /dev/sda` ?

killmasta93, 2020-05-31:
Thanks for the reply this is what i get 

```
root@prometheus:~# smartctl -A -i /dev/sda
smartctl 7.1 2019-12-30 r5022 [x86_64-linux-5.4.34-1-pve] (local build)
Copyright (C) 2002-19, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Vendor:               HP
Product:              EG0300FBVFL
Revision:             HPDC
Compliance:           SPC-3
User Capacity:        300,000,000,000 bytes [300 GB]
Logical block size:   512 bytes
Rotation Rate:        10020 rpm
Form Factor:          2.5 inches
Logical Unit id:      0x5000cca0546795b0
Serial number:        KFHUZ06Z
Device type:          disk
Transport protocol:   SAS (SPL-3)
Local Time is:        Sun May 31 11:54:18 2020 -05
SMART support is:     Available - device has SMART capability.
SMART support is:     Enabled
Temperature Warning:  Enabled

=== START OF READ SMART DATA SECTION ===
Elements in grown defect list: 0
```

nobodysu, 2020-05-31:
What about `# smartctl -a -i /dev/sda` ?

killmasta93, 2020-05-31:
thanks for the reply i got this

```
root@prometheus:~# smartctl -a -i /dev/sda 
smartctl 7.1 2019-12-30 r5022 [x86_64-linux-5.4.34-1-pve] (local build)
Copyright (C) 2002-19, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Vendor:               HP
Product:              EG0300FBVFL
Revision:             HPDC
Compliance:           SPC-3
User Capacity:        300,000,000,000 bytes [300 GB]
Logical block size:   512 bytes
Rotation Rate:        10020 rpm
Form Factor:          2.5 inches
Logical Unit id:      0x5000cca0546795b0
Serial number:        KFHUZ06Z
Device type:          disk
Transport protocol:   SAS (SPL-3)
Local Time is:        Sun May 31 12:38:24 2020 -05
SMART support is:     Available - device has SMART capability.
SMART support is:     Enabled
Temperature Warning:  Enabled

=== START OF READ SMART DATA SECTION ===
SMART Health Status: OK
Current Drive Temperature:     0 C
Drive Trip Temperature:        0 C

Elements in grown defect list: 0

Error Counter logging not supported

Device does not support Self Test logging
```

nobodysu, 2020-05-31:
Hm, but drive does not report the temperature. What about others?

killmasta93, 2020-05-31:
same thing the other drives also could it be the bios settings?

nobodysu, 2020-05-31:
But `sdg` reports correctly. Maybe storage controller is involved here.

killmasta93, 2020-05-31:
hmm, currently running this card

Genuine LSI 6Gbps SAS HBA LSI 9201-8i (=9211-8i) P20 IT Mode ZFS FreeNAS unRAID

nobodysu, 2020-05-31:
Maybe some settings present on that?

killmasta93, 2020-06-10:
i was taking look at the items and found this
`
Cannot evaluate expression: "Cannot evaluate function "prometheusaa:mini.disk.tempCrit[sdb].last()": not enough data.".`

nobodysu, 2020-06-10:
That's because there are no temperature for sdb.