nobody43/zabbix-mini-IPMI #54: Not showing temperature  on SSD disks

killmasta93, 2020-06-10:
**Describe the bug**
Hi there, Currently i have a dell r720 with SSD disks but not showing the temperature but shows the CPU temps
**To Reproduce**
 zabbix_get -s myip -k mini.disktemp.discovery[getverb,"prometheusdellhagroup"]
ZBX_NOTSUPPORTED: Unsupported item key.

```
root@prometheus:~#  smartctl -a -i /dev/sda 
smartctl 6.6 2016-05-31 r4324 [x86_64-linux-4.15.18-12-pve] (local build)
Copyright (C) 2002-16, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Family:     Intel 730 and DC S35x0/3610/3700 Series SSDs
Device Model:     INTEL SSDSC2BB800G4
Serial Number:    PHWL522402FJ800RGN
LU WWN Device Id: 5 5cd2e4 04c55ea1a
Firmware Version: D2010370
User Capacity:    800,166,076,416 bytes [800 GB]
Sector Sizes:     512 bytes logical, 4096 bytes physical
Rotation Rate:    Solid State Device
Form Factor:      2.5 inches
Device is:        In smartctl database [for details use: -P show]
ATA Version is:   ACS-2 T13/2015-D revision 3
SATA Version is:  SATA 2.6, 6.0 Gb/s (current: 6.0 Gb/s)
Local Time is:    Tue Jun  9 21:33:13 2020 -05
SMART support is: Available - device has SMART capability.
SMART support is: Enabled

=== START OF READ SMART DATA SECTION ===
SMART overall-health self-assessment test result: PASSED

General SMART Values:
Offline data collection status:  (0x00)	Offline data collection activity
					was never started.
					Auto Offline Data Collection: Disabled.
Self-test execution status:      (   0)	The previous self-test routine completed
					without error or no self-test has ever 
					been run.
Total time to complete Offline 
data collection: 		(    0) seconds.
Offline data collection
capabilities: 			 (0x79) SMART execute Offline immediate.
					No Auto Offline data collection support.
					Suspend Offline collection upon new
					command.
					Offline surface scan supported.
					Self-test supported.
					Conveyance Self-test supported.
					Selective Self-test supported.
SMART capabilities:            (0x0003)	Saves SMART data before entering
					power-saving mode.
					Supports SMART auto save timer.
Error logging capability:        (0x01)	Error logging supported.
					General Purpose Logging supported.
Short self-test routine 
recommended polling time: 	 (   1) minutes.
Extended self-test routine
recommended polling time: 	 (   2) minutes.
Conveyance self-test routine
recommended polling time: 	 (   2) minutes.
SCT capabilities: 	       (0x003d)	SCT Status supported.
					SCT Error Recovery Control supported.
					SCT Feature Control supported.
					SCT Data Table supported.

SMART Attributes Data Structure revision number: 1
Vendor Specific SMART Attributes with Thresholds:
ID# ATTRIBUTE_NAME          FLAG     VALUE WORST THRESH TYPE      UPDATED  WHEN_FAILED RAW_VALUE
  5 Reallocated_Sector_Ct   0x0032   100   100   000    Old_age   Always       -       2
  9 Power_On_Hours          0x0032   100   100   000    Old_age   Always       -       36320
 12 Power_Cycle_Count       0x0032   100   100   000    Old_age   Always       -       17
170 Available_Reservd_Space 0x0033   099   099   010    Pre-fail  Always       -       0
171 Program_Fail_Count      0x0032   100   100   000    Old_age   Always       -       4
172 Erase_Fail_Count        0x0032   100   100   000    Old_age   Always       -       0
174 Unsafe_Shutdown_Count   0x0032   100   100   000    Old_age   Always       -       12
175 Power_Loss_Cap_Test     0x0033   100   100   010    Pre-fail  Always       -       640 (215 1778)
183 SATA_Downshift_Count    0x0032   100   100   000    Old_age   Always       -       0
184 End-to-End_Error        0x0033   100   100   090    Pre-fail  Always       -       0
187 Reported_Uncorrect      0x0032   100   100   000    Old_age   Always       -       0
190 Temperature_Case        0x0022   085   076   000    Old_age   Always       -       15 (Min/Max 13/24)
192 Unsafe_Shutdown_Count   0x0032   100   100   000    Old_age   Always       -       12
194 Temperature_Internal    0x0022   100   100   000    Old_age   Always       -       26
197 Current_Pending_Sector  0x0032   100   100   000    Old_age   Always       -       0
199 CRC_Error_Count         0x003e   100   100   000    Old_age   Always       -       0
225 Host_Writes_32MiB       0x0032   100   100   000    Old_age   Always       -       2792434
226 Workld_Media_Wear_Indic 0x0032   100   100   000    Old_age   Always       -       8222
227 Workld_Host_Reads_Perc  0x0032   100   100   000    Old_age   Always       -       77
228 Workload_Minutes        0x0032   100   100   000    Old_age   Always       -       2179026
232 Available_Reservd_Space 0x0033   099   099   010    Pre-fail  Always       -       0
233 Media_Wearout_Indicator 0x0032   092   092   000    Old_age   Always       -       0
234 Thermal_Throttle        0x0032   100   100   000    Old_age   Always       -       0/0
241 Host_Writes_32MiB       0x0032   100   100   000    Old_age   Always       -       2792434
242 Host_Reads_32MiB        0x0032   100   100   000    Old_age   Always       -       9647272

SMART Error Log Version: 1
No Errors Logged
```

**Screenshots**
[image]
[image]

**Please complete the following information:**
 - OS: [e.g. Debian] 9
 - Zabbix server version: [e.g. 3.0] 4.4
 - Active or passive check: [e.g. Passive] active
 - Using zabbix proxy: [e.g. No proxy] none

nobodysu, 2020-06-10:
`ZBX_NOTSUPPORTED: Unsupported item key.`
Recheck configuration and restart the agent.

killmasta93, 2020-06-11:
correct i restarted the agent same still shows on zabbix no temperature

nobodysu, 2020-06-11:
Given your configuration this output is impossible:
>root@monitor:~# zabbix_get -s myip -k mini.cputemp.discovery[get,"prometheusdellhagroup"]
ZBX_NOTSUPPORTED: Unsupported item key.

I think there's some host confusion.

killmasta93, 2020-06-11:
Thanks for the reply, im putting the IP of the zabbix agent prometheusdellhagroup

or is it the other way around?

nobodysu, 2020-06-11:
```root@monitor:~# zabbix_get -s myip -k mini.disktemp.discovery[getverb,"prometheusdellhagroup"]```

nobodysu, 2020-06-11:
`root@prometheus:~#  smartctl -A -i /dev/sda`
?

killmasta93, 2020-06-12:
thanks for the reply

```
root@prometheus:~# smartctl -A -i /dev/sda
smartctl 6.6 2016-05-31 r4324 [x86_64-linux-4.15.18-12-pve] (local build)
Copyright (C) 2002-16, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Family:     Intel 730 and DC S35x0/3610/3700 Series SSDs
Device Model:     INTEL SSDSC2BB800G4
Serial Number:    PHWL522402FJ800RGN
LU WWN Device Id: 5 5cd2e4 04c55ea1a
Firmware Version: D2010370
User Capacity:    800,166,076,416 bytes [800 GB]
Sector Sizes:     512 bytes logical, 4096 bytes physical
Rotation Rate:    Solid State Device
Form Factor:      2.5 inches
Device is:        In smartctl database [for details use: -P show]
ATA Version is:   ACS-2 T13/2015-D revision 3
SATA Version is:  SATA 2.6, 6.0 Gb/s (current: 6.0 Gb/s)
Local Time is:    Thu Jun 11 23:34:09 2020 -05
SMART support is: Available - device has SMART capability.
SMART support is: Enabled

=== START OF READ SMART DATA SECTION ===
SMART Attributes Data Structure revision number: 1
Vendor Specific SMART Attributes with Thresholds:
ID# ATTRIBUTE_NAME          FLAG     VALUE WORST THRESH TYPE      UPDATED  WHEN_FAILED RAW_VALUE
  5 Reallocated_Sector_Ct   0x0032   100   100   000    Old_age   Always       -       2
  9 Power_On_Hours          0x0032   100   100   000    Old_age   Always       -       36370
 12 Power_Cycle_Count       0x0032   100   100   000    Old_age   Always       -       17
170 Available_Reservd_Space 0x0033   099   099   010    Pre-fail  Always       -       0
171 Program_Fail_Count      0x0032   100   100   000    Old_age   Always       -       4
172 Erase_Fail_Count        0x0032   100   100   000    Old_age   Always       -       0
174 Unsafe_Shutdown_Count   0x0032   100   100   000    Old_age   Always       -       12
175 Power_Loss_Cap_Test     0x0033   100   100   010    Pre-fail  Always       -       640 (215 4764)
183 SATA_Downshift_Count    0x0032   100   100   000    Old_age   Always       -       0
184 End-to-End_Error        0x0033   100   100   090    Pre-fail  Always       -       0
187 Reported_Uncorrect      0x0032   100   100   000    Old_age   Always       -       0
190 Temperature_Case        0x0022   086   076   000    Old_age   Always       -       14 (Min/Max 13/24)
192 Unsafe_Shutdown_Count   0x0032   100   100   000    Old_age   Always       -       12
194 Temperature_Internal    0x0022   100   100   000    Old_age   Always       -       25
197 Current_Pending_Sector  0x0032   100   100   000    Old_age   Always       -       0
199 CRC_Error_Count         0x003e   100   100   000    Old_age   Always       -       0
225 Host_Writes_32MiB       0x0032   100   100   000    Old_age   Always       -       2792626
226 Workld_Media_Wear_Indic 0x0032   100   100   000    Old_age   Always       -       8222
227 Workld_Host_Reads_Perc  0x0032   100   100   000    Old_age   Always       -       77
228 Workload_Minutes        0x0032   100   100   000    Old_age   Always       -       2182012
232 Available_Reservd_Space 0x0033   099   099   010    Pre-fail  Always       -       0
233 Media_Wearout_Indicator 0x0032   092   092   000    Old_age   Always       -       0
234 Thermal_Throttle        0x0032   100   100   000    Old_age   Always       -       0/0
241 Host_Writes_32MiB       0x0032   100   100   000    Old_age   Always       -       2792626
242 Host_Reads_32MiB        0x0032   100   100   000    Old_age   Always       -       9647272

```
```
root@monitor:~# zabbix_get -s 181.xx.xxx.xx -k mini.disktemp.discovery[getverb,"prometheusdellhagroup"]
ZBX_NOTSUPPORTED: Unsupported item key.

```

nobodysu, 2020-06-12:
Try this one:
https://github.com/nobodysu/zabbix-mini-IPMI/blob/disk_temp_regexps/mini_ipmi_smartctl.py
>root@monitor:~# zabbix_get -s 181.xx.xxx.xx -k mini.disktemp.discovery[getverb,"prometheusdellhagroup"]
ZBX_NOTSUPPORTED: Unsupported item key.

That's impossible. Either 181.xx.xxx.xx is not the host you need or it's incorrectly configured (userparameters).

killmasta93, 2020-06-13:
Thanks for the reply, that update seemed to work but also i know why it was not working because i had to add a -p because i dont use the 10050

```
 zabbix_get -s 181.xx.xx.xx -p 30063 -k mini.disktemp.discovery[getverb,"prometheusdellhagroup"]
  Path guess: linux

  Python version:
 3.5.3 (default, Sep 27 2018, 17:25:39)
[GCC 6.3.0 20170516]

  Sender version:
 zabbix_sender (Zabbix) 3.0.7
Revision 64609 21 December 2016, compilation time: Jun  4 2017 07:14:06

Copyright (C) 2016 Zabbix SIA
License GPLv2+: GNU GPL version 2 or later <http://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it according to
the license. There is NO WARRANTY, to the extent permitted by law.

  Config's main settings:
Server=monitor.mydomain.com
ServerActive=monitor.mydomain.com
Timeout=30

  Please report any issues or missing features to:
https://github.com/nobodysu/zabbix-mini-IPMI/issues

zabbix_sender [38746]: DEBUG: answer [{"response":"success","info":"processed: 42; failed: 0; total: 42; seconds spent: 0.000940"}]
info from server: "processed: 42; failed: 0; total: 42; seconds spent: 0.000940"
sent: 42; skipped: 0; total: 42

  Note: the sender will fail if server did not gather LLD previously.

  Data sent to zabbix sender:

"prometheusdellhagroup" mini.disk.info[sda,DriveStatus] "PROCESSED"
"prometheusdellhagroup" mini.disk.temp[sda] "28"
"prometheusdellhagroup" mini.disk.tempMin[sda] "25"
"prometheusdellhagroup" mini.disk.tempMax[sda] "45"
"prometheusdellhagroup" mini.disk.tempCrit[sda] "60"
"prometheusdellhagroup" mini.disk.info[sdb,DriveStatus] "PROCESSED"
"prometheusdellhagroup" mini.disk.temp[sdb] "34"
"prometheusdellhagroup" mini.disk.tempMin[sdb] "25"
"prometheusdellhagroup" mini.disk.tempMax[sdb] "45"
"prometheusdellhagroup" mini.disk.tempCrit[sdb] "60"
"prometheusdellhagroup" mini.disk.info[sdc,DriveStatus] "PROCESSED"
"prometheusdellhagroup" mini.disk.temp[sdc] "29"
"prometheusdellhagroup" mini.disk.tempMin[sdc] "25"
"prometheusdellhagroup" mini.disk.tempMax[sdc] "45"
"prometheusdellhagroup" mini.disk.tempCrit[sdc] "60"
"prometheusdellhagroup" mini.disk.info[sdd,DriveStatus] "PROCESSED"
"prometheusdellhagroup" mini.disk.temp[sdd] "28"
"prometheusdellhagroup" mini.disk.tempMin[sdd] "25"
"prometheusdellhagroup" mini.disk.tempMax[sdd] "45"
"prometheusdellhagroup" mini.disk.tempCrit[sdd] "60"
"prometheusdellhagroup" mini.disk.info[sde,DriveStatus] "PROCESSED"
"prometheusdellhagroup" mini.disk.temp[sde] "31"
"prometheusdellhagroup" mini.disk.tempMin[sde] "25"
"prometheusdellhagroup" mini.disk.tempMax[sde] "45"
"prometheusdellhagroup" mini.disk.tempCrit[sde] "60"
"prometheusdellhagroup" mini.disk.info[sdf,DriveStatus] "PROCESSED"
"prometheusdellhagroup" mini.disk.temp[sdf] "28"
"prometheusdellhagroup" mini.disk.tempMin[sdf] "25"
"prometheusdellhagroup" mini.disk.tempMax[sdf] "45"
"prometheusdellhagroup" mini.disk.tempCrit[sdf] "60"
"prometheusdellhagroup" mini.disk.info[sdg,DriveStatus] "PROCESSED"
"prometheusdellhagroup" mini.disk.temp[sdg] "30"
"prometheusdellhagroup" mini.disk.tempMin[sdg] "25"
"prometheusdellhagroup" mini.disk.tempMax[sdg] "45"
"prometheusdellhagroup" mini.disk.tempCrit[sdg] "60"
"prometheusdellhagroup" mini.disk.info[sdh,DriveStatus] "PROCESSED"
"prometheusdellhagroup" mini.disk.temp[sdh] "29"
"prometheusdellhagroup" mini.disk.tempMin[sdh] "25"
"prometheusdellhagroup" mini.disk.tempMax[sdh] "45"
"prometheusdellhagroup" mini.disk.tempCrit[sdh] "60"
"prometheusdellhagroup" mini.disk.info[ConfigStatus] "CONFIGURED"
"prometheusdellhagroup" mini.disk.temp[MAX] "34"
```

killmasta93, 2020-06-16:
Hi there again, i was checking and found this error 

[image]

nobodysu, 2020-06-16:
`zabbix_get -s 192.0.2.1 -k mini.disktemp.discovery[getverb,"Example host"]`

killmasta93, 2020-06-16:
```
root@monitor:~# zabbix_get -s 181.xx,xx.xx -p 30063 -k mini.disktemp.discovery[getverb,"prometheusdellhagroup"]
  Path guess: linux

  Python version:
 3.5.3 (default, Sep 27 2018, 17:25:39) 
[GCC 6.3.0 20170516]

  Sender version:
 zabbix_sender (Zabbix) 3.0.7
Revision 64609 21 December 2016, compilation time: Jun  4 2017 07:14:06

Copyright (C) 2016 Zabbix SIA
License GPLv2+: GNU GPL version 2 or later <http://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it according to
the license. There is NO WARRANTY, to the extent permitted by law.

  Config's main settings:
Server=monitor.mydomain.com
ServerActive=monitor.mydomain.com
Timeout=30

  Please report any issues or missing features to:
https://github.com/nobodysu/zabbix-mini-IPMI/issues

zabbix_sender [28515]: DEBUG: answer [{"response":"success","info":"processed: 42; failed: 0; total: 42; seconds spent: 0.000918"}]
info from server: "processed: 42; failed: 0; total: 42; seconds spent: 0.000918"
sent: 42; skipped: 0; total: 42

  Note: the sender will fail if server did not gather LLD previously.

  Data sent to zabbix sender:

"prometheusdellhagroup" mini.disk.info[sda,DriveStatus] "PROCESSED"
"prometheusdellhagroup" mini.disk.temp[sda] "30"
"prometheusdellhagroup" mini.disk.tempMin[sda] "25"
"prometheusdellhagroup" mini.disk.tempMax[sda] "45"
"prometheusdellhagroup" mini.disk.tempCrit[sda] "60"
"prometheusdellhagroup" mini.disk.info[sdb,DriveStatus] "PROCESSED"
"prometheusdellhagroup" mini.disk.temp[sdb] "35"
"prometheusdellhagroup" mini.disk.tempMin[sdb] "25"
"prometheusdellhagroup" mini.disk.tempMax[sdb] "45"
"prometheusdellhagroup" mini.disk.tempCrit[sdb] "60"
"prometheusdellhagroup" mini.disk.info[sdc,DriveStatus] "PROCESSED"
"prometheusdellhagroup" mini.disk.temp[sdc] "30"
"prometheusdellhagroup" mini.disk.tempMin[sdc] "25"
"prometheusdellhagroup" mini.disk.tempMax[sdc] "45"
"prometheusdellhagroup" mini.disk.tempCrit[sdc] "60"
"prometheusdellhagroup" mini.disk.info[sdd,DriveStatus] "PROCESSED"
"prometheusdellhagroup" mini.disk.temp[sdd] "29"
"prometheusdellhagroup" mini.disk.tempMin[sdd] "25"
"prometheusdellhagroup" mini.disk.tempMax[sdd] "45"
"prometheusdellhagroup" mini.disk.tempCrit[sdd] "60"
"prometheusdellhagroup" mini.disk.info[sde,DriveStatus] "PROCESSED"
"prometheusdellhagroup" mini.disk.temp[sde] "32"
"prometheusdellhagroup" mini.disk.tempMin[sde] "25"
"prometheusdellhagroup" mini.disk.tempMax[sde] "45"
"prometheusdellhagroup" mini.disk.tempCrit[sde] "60"
"prometheusdellhagroup" mini.disk.info[sdf,DriveStatus] "PROCESSED"
"prometheusdellhagroup" mini.disk.temp[sdf] "30"
"prometheusdellhagroup" mini.disk.tempMin[sdf] "25"
"prometheusdellhagroup" mini.disk.tempMax[sdf] "45"
"prometheusdellhagroup" mini.disk.tempCrit[sdf] "60"
"prometheusdellhagroup" mini.disk.info[sdg,DriveStatus] "PROCESSED"
"prometheusdellhagroup" mini.disk.temp[sdg] "31"
"prometheusdellhagroup" mini.disk.tempMin[sdg] "25"
"prometheusdellhagroup" mini.disk.tempMax[sdg] "45"
"prometheusdellhagroup" mini.disk.tempCrit[sdg] "60"
"prometheusdellhagroup" mini.disk.info[sdh,DriveStatus] "PROCESSED"
"prometheusdellhagroup" mini.disk.temp[sdh] "30"
"prometheusdellhagroup" mini.disk.tempMin[sdh] "25"
"prometheusdellhagroup" mini.disk.tempMax[sdh] "45"
"prometheusdellhagroup" mini.disk.tempCrit[sdh] "60"
"prometheusdellhagroup" mini.disk.info[ConfigStatus] "CONFIGURED"
"prometheusdellhagroup" mini.disk.temp[MAX] "35"

```

nobodysu, 2020-06-17:
That should work. What about
`zabbix_get -s 192.0.2.1 -k mini.disktemp.discovery[get,"Example host"]`
?

killmasta93, 2020-06-17:
thanks for the reply this is what i get

```
root@monitor:~# zabbix_get -s 181.xx.xx.xx -p 30063 -k mini.disktemp.discovery[get,"prometheusdellhagroup"]
{
    "data": [
        {
            "{#DISK}": "sda"
        },
        {
            "{#DISK}": "sdb"
        },
        {
            "{#DISK}": "sdc"
        },
        {
            "{#DISK}": "sdd"
        },
        {
            "{#DISK}": "sde"
        },
        {
            "{#DISK}": "sdf"
        },
        {
            "{#DISK}": "sdg"
        },
        {
            "{#DISK}": "sdh"
        }
    ]
}
```
But now on the console shows the temp of the disk , whats odd is the error on the discovery and only 2 disks cannot seem to show temperature

[image]

[image]

the two disks sdi and sdj
but the temperature info show on the S.M.A.R.T

```
root@prometheus:~# smartctl -A -i /dev/sdi
smartctl 6.6 2016-05-31 r4324 [x86_64-linux-4.15.18-12-pve] (local build)
Copyright (C) 2002-16, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Family:     Intel 730 and DC S35x0/3610/3700 Series SSDs
Device Model:     INTEL SSDSC2BB800G4
Serial Number:    CVWL4205025H800RGN
LU WWN Device Id: 5 5cd2e4 04bc630be
Firmware Version: D2010370
User Capacity:    800,166,076,416 bytes [800 GB]
Sector Sizes:     512 bytes logical, 4096 bytes physical
Rotation Rate:    Solid State Device
Form Factor:      2.5 inches
Device is:        In smartctl database [for details use: -P show]
ATA Version is:   ACS-2 T13/2015-D revision 3
SATA Version is:  SATA 2.6, 6.0 Gb/s (current: 6.0 Gb/s)
Local Time is:    Wed Jun 17 14:33:19 2020 -05
SMART support is: Available - device has SMART capability.
SMART support is: Enabled

=== START OF READ SMART DATA SECTION ===
SMART Attributes Data Structure revision number: 1
Vendor Specific SMART Attributes with Thresholds:
ID# ATTRIBUTE_NAME          FLAG     VALUE WORST THRESH TYPE      UPDATED  WHEN_FAILED RAW_VALUE
  5 Reallocated_Sector_Ct   0x0032   100   100   000    Old_age   Always       -       0
  9 Power_On_Hours          0x0032   100   100   000    Old_age   Always       -       40545
[... 55 lines cut ...]
171 Program_Fail_Count      0x0032   100   100   000    Old_age   Always       -       0
172 Erase_Fail_Count        0x0032   100   100   000    Old_age   Always       -       0
174 Unsafe_Shutdown_Count   0x0032   100   100   000    Old_age   Always       -       14
175 Power_Loss_Cap_Test     0x0033   100   100   010    Pre-fail  Always       -       657 (220 2744)
183 SATA_Downshift_Count    0x0032   100   100   000    Old_age   Always       -       0
184 End-to-End_Error        0x0033   100   100   090    Pre-fail  Always       -       0
187 Reported_Uncorrect      0x0032   100   100   000    Old_age   Always       -       0
190 Temperature_Case        0x0022   084   077   000    Old_age   Always       -       16 (Min/Max 12/25)
192 Unsafe_Shutdown_Count   0x0032   100   100   000    Old_age   Always       -       14
194 Temperature_Internal    0x0022   100   100   000    Old_age   Always       -       27
197 Current_Pending_Sector  0x0032   100   100   000    Old_age   Always       -       0
199 CRC_Error_Count         0x003e   100   100   000    Old_age   Always       -       0
225 Host_Writes_32MiB       0x0032   100   100   000    Old_age   Always       -       2732864
226 Workld_Media_Wear_Indic 0x0032   100   100   000    Old_age   Always       -       7065
227 Workld_Host_Reads_Perc  0x0032   100   100   000    Old_age   Always       -       79
228 Workload_Minutes        0x0032   100   100   000    Old_age   Always       -       2225513
232 Available_Reservd_Space 0x0033   100   100   010    Pre-fail  Always       -       0
233 Media_Wearout_Indicator 0x0032   094   094   000    Old_age   Always       -       0
234 Thermal_Throttle        0x0032   100   100   000    Old_age   Always       -       0/0
241 Host_Writes_32MiB       0x0032   100   100   000    Old_age   Always       -       2732864
242 Host_Reads_32MiB        0x0032   100   100   000    Old_age   Always       -       10399520

```

Thank you again

nobodysu, 2020-06-18:
Hm, is that [nvme](https://github.com/nobodysu/zabbix-mini-IPMI/blob/master/mini_ipmi_smartctl.py#L27)? They are disabled by default. If not resolved - please provide:
`smartctl --scan`
`smartctl --scan -d nvme`

killmasta93, 2020-06-19:
Thanks for the reply, currently there intel SSD 2.5 
```

root@prometheus:~# smartctl --scan
/dev/sda -d scsi # /dev/sda, SCSI device
/dev/sdb -d scsi # /dev/sdb, SCSI device
/dev/sdc -d scsi # /dev/sdc, SCSI device
/dev/sdd -d scsi # /dev/sdd, SCSI device
/dev/sde -d scsi # /dev/sde, SCSI device
/dev/sdf -d scsi # /dev/sdf, SCSI device
/dev/sdg -d scsi # /dev/sdg, SCSI device
/dev/sdh -d scsi # /dev/sdh, SCSI device
/dev/sdi -d scsi # /dev/sdi, SCSI device
/dev/sdj -d scsi # /dev/sdj, SCSI device
root@prometheus:~# smartctl --scan -d nvme
# scan_smart_devices: glob(3) aborted matching pattern /dev/discs/disc*

```
```
root@prometheus:~# smartctl -A -i /dev/sdj
smartctl 6.6 2016-05-31 r4324 [x86_64-linux-4.15.18-12-pve] (local build)
Copyright (C) 2002-16, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Family:     Intel 730 and DC S35x0/3610/3700 Series SSDs
Device Model:     INTEL SSDSC2BB800G4
Serial Number:    PHWL503402WQ800RGN
LU WWN Device Id: 5 5cd2e4 04c4c0d90
Firmware Version: D2010370
User Capacity:    800,166,076,416 bytes [800 GB]
Sector Sizes:     512 bytes logical, 4096 bytes physical
Rotation Rate:    Solid State Device
Form Factor:      2.5 inches
Device is:        In smartctl database [for details use: -P show]
ATA Version is:   ACS-2 T13/2015-D revision 3
SATA Version is:  SATA 2.6, 6.0 Gb/s (current: 6.0 Gb/s)
Local Time is:    Thu Jun 18 19:47:24 2020 -05
SMART support is: Available - device has SMART capability.
SMART support is: Enabled

```

nobodysu, 2020-06-19:
You are not providing full information. It's either partially old configuration or different computers.
Having these two outputs at the same time is impossible:
`root@monitor:~# zabbix_get -s 181.xx.xx.xx -p 30063 -k mini.disktemp.discovery[get,"prometheusdellhagroup"]`
`root@prometheus:~# smartctl --scan`
Please double check.

killmasta93, 2020-06-19:
Thanks for the reply, sorry for not being clear the root@monitor is my zabbix server which is at another site with another WAN IP  and the root@prometheus is the dell server with the SSDs with another WAN ip and another site.

nobodysu, 2020-07-18:
You could run the following command to determine presence of temperature sensor:
`smartctl -i -l scttemp /dev/sdX`

killmasta93, 2020-07-20:
Thanks for the reply, the update of the template seems to show now

closed 2020-07-20 by killmasta93