nobody43/zabbix-mini-IPMI #57: Changing temp threshold  on only one host?

killmasta93, 2020-07-13:
Hi,
I was wondering if its possible to change the threshold only for one Host? i tried checking on triggers to change but couldn't find the option 

Thank you

nobodysu, 2020-07-13:
Disks or cpus? Linux or Windows?

killmasta93, 2020-07-13:
thanks for the reply, only CPUs

nobodysu, 2020-07-13:
_Host_ -> Macros -> Inherited -> Change

killmasta93, 2020-07-13:
thanks for the reply, i went to that section but did not see anything of mini IPMI temps
[image]

nobodysu, 2020-07-13:
Means template is not assigned to the host.

killmasta93, 2020-07-13:
i checked and it is assigned

[image]

nobodysu, 2020-07-14:
Then template is not up to date:
https://github.com/nobodysu/zabbix-mini-IPMI/blob/master/Template_mini-IPMI_v2.xml

killmasta93, 2020-07-15:
thanks for the reply, i tried updating the template by adding and i get this 
[image]

killmasta93, 2020-07-15:
after checking it did make the changes thank you

closed 2020-07-15 by killmasta93

killmasta93, 2020-07-15:
quick question forgot to mention if i would want to do the same for the disk temp?

nobodysu, 2020-07-15:
Right now it's in the file: https://github.com/nobodysu/zabbix-mini-IPMI/blob/master/mini_ipmi_smartctl.py#L32
Will be moved to macro someday.

killmasta93, 2020-07-16:
Thanks for the reply, so this would apply to all hosts if changing the .py? possible to change on the template for all hosts

nobodysu, 2020-07-16:
On every host there is a copy of `mini_ipmi_smartctl.py` with separate thresholds. Therefore, it's all per-host and cannot be changed through template, yet.
However, I advice you to think twice before changing disk thresholds because it's based on Goggle research: http://www.buildcomputers.net/hdd-temperature.html

killmasta93, 2020-07-16:
Thank you again, your right ill keep that in mind