nobody43/zabbix-mini-IPMI #60: Cpu temp shows double

killmasta93, 2020-07-26:
Hi,
Currently after updating the new template, getting to show double CPU not sure why?
Also on the older  template it would alert me the temperature that it was at 
currently only alert me saying its over 65 C and not showing the alert on the temperature that it is at
Thank you 
[image]

nobodysu, 2020-07-26:
Global trigger is disabled by default in the template: https://github.com/nobodysu/zabbix-mini-IPMI/blob/master/Template_mini-IPMI_v2.xml#L3279

killmasta93, 2020-07-26:
thanks for the quick reply, so this is the reason im not getting the current temperature of the alert?
and what about the double CPU alert?

nobodysu, 2020-07-26:
Just disable `CPU temperature is too high` in the template.

Last value in trigger name is moved to comments because it's a bad practice and can mess up triggers in latest versions of zabbix.

killmasta93, 2020-07-26:
So i would disable one of the cpu temperature is too high, but there is two 
[image]

[image]

nobodysu, 2020-07-26:
First one.

killmasta93, 2020-07-27:
thank you that did the trick so the alert of the  current temperature not possible? or can i enable it on the template?

nobodysu, 2020-07-27:
You could, but it's strongly discouraged - it could bread reporting for these problems. Now `last value` is hidden in problem details.

closed 2020-07-30 by killmasta93