nobody43/zabbix-mini-IPMI #61: Invalid discovery rule value

Adam0097, 2020-08-03:
I m new in this zabbix world and I am trying to upload the template to my test server to verify that it works, I still encounter a problem with "Invalid discovery rule value: cannot parse as a valid JSON object: invalid object format, expected opening character '{' or '[' at: ' 'python' is not recognized as an internal or external command,
operable program or batch file. '"after uploading the template and assigning it to the host, Python is installed, as well as OpenHardwareMonitorReport, smartools as well, scripts placed (with path configure) in the Zabbix Agent folder

[image]

[image]

[image]

 - OS Server: OcentOs OS Agent:Windows
 - Zabbix server version: 5.0
 - Using zabbix proxy: No proxy

nobodysu, 2020-08-03:
Python is not in `PATH` apparently. Here's how it can be solved:
https://github.com/nobodysu/zabbix-mini-IPMI/blob/master/screenshots/python-installation1.png
https://github.com/nobodysu/zabbix-mini-IPMI/blob/master/screenshots/python-installation2.png
Reboot is required afterwards.

Adam0097, 2020-08-03:
in theory, it only managed to throw such errors in the info

[image]

and no data collectin

[image]

nobodysu, 2020-08-03:
These errors are expected and data should be there by now.
Provide outputs if not:
https://github.com/nobodysu/zabbix-mini-IPMI#testing

Adam0097, 2020-08-04:
From client
[image]

Adam0097, 2020-08-04:
Fixed, next error
[image]

nobodysu, 2020-08-04:
Are you running the test from admin console (right mouse button -> run as admin)?
Please post the server query - it will tell more.

Adam0097, 2020-08-04:
The problem seems to be solved, I removed the port adress from ServerActive and it worked !

[image]

closed 2020-08-04 by Adam0097