nobody43/zabbix-mini-IPMI #63: Using the plugin with Zabbix Agent2 on Windows

Fedetiss, 2020-10-16:
**Describe the problem**
Data is only sent when using the test command, and only some data is processed 

**To Reproduce**
I'm using Zabbix Agent2, copied a zabbix_sender.exe from another Zabbix Agent installation. Every path is set according to installation

**Expected behavior**
I want to start getting data on Zabbix

**Provide all outputs described in [Testing](https://github.com/nobodysu/zabbix-mini-IPMI#testing) step**
[image]
[image]

**Please complete the following information:**
 - OS: Windows Server 2008 R2
 - Zabbix server version: 5.0
 - Active or passive check: Active
 - Using zabbix proxy: No Proxy

nobodysu, 2020-10-16:
Have you modified the paths accordingly?
https://github.com/nobodysu/zabbix-mini-IPMI/blob/master/Win/zabbix_agentd.conf.d/userparameter_mini-ipmi2.conf

Fedetiss, 2020-10-19:
> Have you modified the paths accordingly?
> https://github.com/nobodysu/zabbix-mini-IPMI/blob/master/Win/zabbix_agentd.conf.d/userparameter_mini-ipmi2.conf

I've created a zabbix-agent folder to match the path, should I move the files into the zabbix installation folder?

nobodysu, 2020-10-20:
`.conf` files from this repository should be placed in `zabbix_agentd.conf.d` directory, which is mentioned in `zabbix_agentd.conf` as an include. Service restart is required.

Fedetiss, 2020-10-21:
> `.conf` files from this repository should be placed in `zabbix_agentd.conf.d` directory, which is mentioned in `zabbix_agentd.conf` as an include. Service restart is required.

Already tried that, now it won't recognize the path to the script:
[image]

I don't know why is not recognizing the path even though I'm using quotation marks 

In my conf file:

UserParameter=mini.disktemp.discovery[*], python "C:\Program Files\Zabbix Agent 2\scripts\mini_ipmi_smartctl.py" "$1" "$2" 
UserParameter=mini.cputemp.discovery[*], python "C:\Program Files\Zabbix Agent 2\scripts\mini_ipmi_ohmr.py" "$1" "$2"

nobodysu, 2020-10-21:
Strange! I'm using it with these paths on Zabbix 5.
Try to move the `.conf` files to shorter path without spaces and change `zabbix_agentd.conf` to include the `include`.

Fedetiss, 2020-10-21:
> Strange! I'm using it with these paths on Zabbix 5.
> Try to move the `.conf` files to shorter path without spaces and change `zabbix_agentd.conf` to include the `include`.

That's what I was doing on the first images, I had everything on the default C:\zabix-agent folder including it on the zabbix_agent2.conf

nobodysu, 2020-10-21:
What about python version? Special permissions to the "missing" file? Does the SYSTEM account have administrative permissions? Could it access the file?

nobodysu, 2021-08-23:
Hope that's fixed.

closed 2021-08-23 by nobodysu