nobody43/zabbix-mini-IPMI #64: Issue Unsupported item key

killmasta93, 2020-10-18:
**Describe the problem**
Hi currently changed from direct zabbix agent to zabbix proxy and whats odd is that im getting Unsupported item key

**To Reproduce**

on client: 
```
   Config's main settings:
Server=192.168.1.10
ServerActive=192.168.1.10
Timeout=30

  Please report any issues or missing features to:
https://github.com/nobodysu/zabbix-mini-IPMI/issues

root@prometheus:~# 
  Note: the sender will fail if server did not gather LLD previously.

  Data sent to zabbix sender:

"prometheusanquimico" mini.disk.info[sda,DriveStatus] "PROCESSED"
"prometheusanquimico" mini.disk.temp[sda] "34"
"prometheusanquimico" mini.disk.tempMin[sda] "25"
"prometheusanquimico" mini.disk.tempMax[sda] "45"
"prometheusanquimico" mini.disk.tempCrit[sda] "60"
"prometheusanquimico" mini.disk.info[sdb,DriveStatus] "PROCESSED"
"prometheusanquimico" mini.disk.temp[sdb] "33"
"prometheusanquimico" mini.disk.tempMin[sdb] "25"
"prometheusanquimico" mini.disk.tempMax[sdb] "45"
"prometheusanquimico" mini.disk.tempCrit[sdb] "60"
"prometheusanquimico" mini.disk.info[ConfigStatus] "CONFIGURED"
"prometheusanquimico" mini.disk.temp[MAX] "34"
zabbix_sender [9264]: DEBUG: answer [{"response":"success","info":"processed: 12; failed: 0; total: 12; seconds spent: 0.000133"}]
info from server: "processed: 12; failed: 0; total: 12; seconds spent: 0.000133"
sent: 12; skipped: 0; total: 12

```

on Server not sure whats the command because the agent is using a proxy

**Please complete the following information:**
 - OS: [e.g. Debian] debian 9
 - Zabbix server version: [e.g. 3.0] 5.0
 - Active or passive check: [e.g. Passive] Active
 - Using zabbix proxy: [e.g. No proxy] using proxy

nobodysu, 2020-10-21:
`processed: 12; failed: 0`
Hm?
Is the host configured to be processed by proxy?

killmasta93, 2020-10-22:
thanks for the reply, correct i have the agents point to the proxy which then the proxy pointed to the zabbix server

nobodysu, 2020-10-22:
Please post all screenshots with all errors including server log.

killmasta93, 2020-10-23:
seems that i need to install also on the proxy host the config files for it to work

killmasta93, 2020-11-01:
so i thought it was working but when i checked the lasted data i got this 

[image]

nobodysu, 2020-11-03:
`client# ls -l /etc/zabbix/scripts/`
`client# ls -l /etc/zabbix/zabbix_agentd.d/`
`client# grep ^Include /etc/zabbix/zabbix_agentd.conf`
`proxy$ zabbix_get -s 192.0.2.1 -k mini.disktemp.discovery[get,"Example host"]`
\+ screenshot from host's Discovery and proxy+server log.

killmasta93, 2020-11-03:
Thanks for the reply,
on the client side
```
root@prometheus4:~# ls -l /etc/zabbix/scripts/
total 49
-rwxr-xr-x 1 root zabbix  9699 Dec 19  2019 mini_ipmi_lmsensors.py
-rwxr-xr-x 1 root zabbix 11719 Dec 19  2019 mini_ipmi_smartctl.py
-rwxr-xr-x 1 root root    2568 Mar 28  2020 proxmox-vms-backup-status
-rwxr-xr-x 1 root root    1645 Mar 28  2020 proxmox-vms-discovery
-rwxr-xr-x 1 root root     885 Jun 22 12:57 pve-zsync-discovery.pl
-rwxr-xr-x 1 root root    4903 Jun 22 12:57 pve-zsync-status.py
drwxr-xr-x 2 root root       3 Dec 19  2019 __pycache__
-rwxr-xr-x 1 root zabbix  7110 Dec 19  2019 sender_wrapper.py
-rwxr-xr-x 1 root zabbix 16817 Dec  6  2019 smartctl-lld.py
root@prometheus4:~# ls -l /etc/zabbix/zabbix_agentd.d
-rw-r--r-- 1 root root 184 Mar 27  2020 /etc/zabbix/zabbix_agentd.d
root@prometheus4:~# grep ^Include /etc/zabbix/zabbix_agentd.conf
Include=/etc/zabbix/zabbix_agentd.conf.d/*.conf
```

the proxy 

```
root@cloud:~# zabbix_get -s 192.168.1.250 -k mini.disktemp.discovery[get,"prometheus4anquimico"]
{
    "data": [
        {
            "{#DISK}": "sda"
        },
        {
            "{#DISK}": "sdb"
        }
    ]
}

```
```
root@cloud:~# zabbix_get -s 192.168.1.250 -k mini.cputemp.discovery[get,"prometheus4anquimico"]
{
    "data": [
        {
            "{#CPU}": 0
        },
        {
            "{#CORE}": "0",
            "{#CPUC}": 0
        },
        {
            "{#CORE}": "1",
            "{#CPUC}": 0
        },
        {
            "{#CORE}": "2",
            "{#CPUC}": 0
        },
        {
            "{#CORE}": "3",
            "{#CPUC}": 0
        },
        {
            "{#CORE}": "4",
            "{#CPUC}": 0
        },
        {
            "{#CORE}": "5",
            "{#CPUC}": 0
        }
    ]
}

```

nobodysu, 2020-11-03:
`host -> Discovery rules` screenshot?
Host configuration screenshot?
Proxy health?

killmasta93, 2020-11-03:
[image]
[image]

[image]

nobodysu, 2020-11-03:
`client# grep ^Server /etc/zabbix/zabbix_agentd.conf`
`client$ ip a`
`proxy$ ip a`

killmasta93, 2020-11-03:
this is the client:

```
root@prometheus4:~# grep ^Server /etc/zabbix/zabbix_agentd.conf
Server=192.168.1.10
ServerActive=192.168.1.10
root@prometheus4:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: enp2s0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 50:3e:aa:08:c7:29 brd ff:ff:ff:ff:ff:ff
3: enp3s0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 50:3e:aa:08:7b:0d brd ff:ff:ff:ff:ff:ff
4: enp4s0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast master vmbr0 state UP group default qlen 1000
    link/ether 4c:ed:fb:3e:b9:b4 brd ff:ff:ff:ff:ff:ff
5: vmbr0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 4c:ed:fb:3e:b9:b4 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.250/24 brd 192.168.1.255 scope global vmbr0
       valid_lft forever preferred_lft forever
    inet6 fe80::4eed:fbff:fe3e:b9b4/64 scope link 
       valid_lft forever preferred_lft forever
6: tap100i0: <BROADCAST,MULTICAST,PROMISC,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast master vmbr0 state UNKNOWN group default qlen 1000
    link/ether a6:46:7a:26:20:c3 brd ff:ff:ff:ff:ff:ff
7: tap101i0: <BROADCAST,MULTICAST,PROMISC,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast master vmbr0 state UNKNOWN group default qlen 1000
    link/ether 06:31:2c:27:ad:eb brd ff:ff:ff:ff:ff:ff
8: tap103i0: <BROADCAST,MULTICAST,PROMISC,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast master vmbr0 state UNKNOWN group default qlen 1000
    link/ether e6:24:82:80:ea:ae brd ff:ff:ff:ff:ff:ff
```

and the proxy

```
root@cloud:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: ens18: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether de:bd:85:97:20:57 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.10/24 brd 192.168.1.255 scope global ens18
       valid_lft forever preferred_lft forever
    inet6 fe80::dcbd:85ff:fe97:2057/64 scope link 
       valid_lft forever preferred_lft forever
```

nobodysu, 2020-11-03:
The solution is to put `192.168.1.250` in Agent `IP address` field.

killmasta93, 2020-11-03:
thanks for the reply, but i got lost, didnt understand. put the 192.168.1.250 in which config?

killmasta93, 2020-11-03:
o wait put 192.168.1.250 instead of 127.0.0.1 let me try and post back

killmasta93, 2020-11-04:
correct that fixed the issue

closed 2020-11-04 by killmasta93