nobody43/zabbix-mini-IPMI #67: "Unsupported item key" after upgrading from zabbix 5.0. to 5.4

packy, 2021-06-13:
**Describe the problem**
Upgraded server from FreeBSD 12.1 to 12.2; zabbix web interface broke. Upgraded zabbix from 5.0 to 5.4. Web interface started working, but no longer able to monitor the server's temps.  I get "Unsupported item key" when I try to use zabbix_get. 

**To Reproduce**
On server/client:
```
✔ daffy 11:26 # zabbix_get -s 192.168.1.11 -k mini.cputemp.discovery[get,"daffy"]
ZBX_NOTSUPPORTED: Unsupported item key.
✔ daffy 11:34 # /usr/local/etc/zabbix/scripts/mini_ipmi_bsdcpu.py getverb daffy
  Python version:
 3.7.9 (default, Feb  4 2021, 01:17:36)
[Clang 10.0.1 (git@github.com:llvm/llvm-project.git llvmorg-10.0.1-0-gef32c611a

  Sender version:
 zabbix_sender (Zabbix) 5.4.1rc1
Revision {ZABBIX_REVISION} 14 May 2021, compilation time: May 20 2021 21:32:24

Copyright (C) 2021 Zabbix SIA
License GPLv2+: GNU GPL version 2 or later <http://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it according to
the license. There is NO WARRANTY, to the extent permitted by law.

This product includes software developed by the OpenSSL Project
for use in the OpenSSL Toolkit (http://www.openssl.org/).

Compiled with OpenSSL 1.1.1h-freebsd  22 Sep 2020
Running with OpenSSL 1.1.1h-freebsd  22 Sep 2020

  Config's main settings:
Server=192.168.1.11,127.0.0.1
ServerActive=192.168.1.11,127.0.0.1
Timeout = 30

  Please report any issues or missing features to:
https://github.com/nobodysu/zabbix-mini-IPMI/issues

✔ daffy 11:36 #
  Note: the sender will fail if server did not gather LLD previously.

  Data sent to zabbix sender:

"daffy" mini.cpu.temp[cpu0,core31] "56"
"daffy" mini.cpu.temp[cpu0,core30] "55"
"daffy" mini.cpu.temp[cpu0,core29] "61"
"daffy" mini.cpu.temp[cpu0,core28] "61"
"daffy" mini.cpu.temp[cpu0,core27] "57"
"daffy" mini.cpu.temp[cpu0,core26] "57"
"daffy" mini.cpu.temp[cpu0,core25] "56"
"daffy" mini.cpu.temp[cpu0,core24] "56"
"daffy" mini.cpu.temp[cpu0,core23] "58"
"daffy" mini.cpu.temp[cpu0,core22] "59"
"daffy" mini.cpu.temp[cpu0,core21] "58"
"daffy" mini.cpu.temp[cpu0,core20] "58"
"daffy" mini.cpu.temp[cpu0,core19] "55"
"daffy" mini.cpu.temp[cpu0,core18] "55"
"daffy" mini.cpu.temp[cpu0,core17] "55"
"daffy" mini.cpu.temp[cpu0,core16] "55"
"daffy" mini.cpu.temp[cpu0,core15] "74"
"daffy" mini.cpu.temp[cpu0,core14] "74"
"daffy" mini.cpu.temp[cpu0,core13] "75"
"daffy" mini.cpu.temp[cpu0,core12] "75"
"daffy" mini.cpu.temp[cpu0,core11] "76"
"daffy" mini.cpu.temp[cpu0,core10] "76"
"daffy" mini.cpu.temp[cpu0,core9] "76"
"daffy" mini.cpu.temp[cpu0,core8] "76"
"daffy" mini.cpu.temp[cpu0,core7] "73"
"daffy" mini.cpu.temp[cpu0,core6] "73"
"daffy" mini.cpu.temp[cpu0,core5] "77"
"daffy" mini.cpu.temp[cpu0,core4] "77"
"daffy" mini.cpu.temp[cpu0,core3] "74"
"daffy" mini.cpu.temp[cpu0,core2] "74"
"daffy" mini.cpu.temp[cpu0,core1] "75"
"daffy" mini.cpu.temp[cpu0,core0] "75"
"daffy" mini.cpu.info[cpu0,TjMax] "70"
"daffy" mini.cpu.temp[cpu0,MAX] "77"
"daffy" mini.cpu.temp[MAX] "77"
"daffy" mini.cpu.info[ConfigStatus] "CONFIGURED"
zabbix_sender [71884]: DEBUG: answer [{"response":"success","info":"processed: 2; failed: 34; total: 36; seconds spent: 0.000182"}]
Response from "192.168.1.11:10051": "processed: 2; failed: 34; total: 36; seconds spent: 0.000182"
zabbix_sender [71885]: DEBUG: answer [{"response":"success","info":"processed: 2; failed: 34; total: 36; seconds spent: 0.000170"}]
Response from "127.0.0.1:10051": "processed: 2; failed: 34; total: 36; seconds spent: 0.000170"
sent: 36; skipped: 0; total: 36
```

**Expected behavior**
I had this working before I upgraded .

**Provide all outputs described in [Testing](https://github.com/nobodysu/zabbix-mini-IPMI#testing) step**
```
✔ daffy 11:41 # zabbix_get -s 192.168.1.11 -k mini.cputemp.discovery[get,"Example host"]
ZBX_NOTSUPPORTED: Unsupported item key.
✔ daffy 11:43 # zabbix_get -s 192.168.1.11 -k mini.disktemp.discovery[get,"Example host"]
ZBX_NOTSUPPORTED: Unsupported item key.
```

**Please complete the following information:**
 - OS: `FreeBSD daffy 12.2-RELEASE-p6 FreeBSD 12.2-RELEASE-p6 GENERIC  amd64`
 - Zabbix server version: 5.4.1rc1
 - Active or passive check: Active
 - Using zabbix proxy: No proxy

**Additional context**
```
✔ daffy 11:38 # ls -l /usr/local/etc/zabbix/
total 179
-rw-r--r--  1 root  wheel   1940 May 20 21:38 libzbxpgsql.conf
drwxr-xr-x  2 root  wheel      3 May 20 21:37 modules
drwxr-xr-x  3 root  wheel      8 May 20 22:18 scripts
-rw-r--r--  1 root  wheel  15599 Jun 11 18:42 zabbix_agentd.conf
drwxr-xr-x  2 root  wheel      2 May 20 21:33 zabbix_agentd.conf.d
-rw-r--r--  1 root  wheel  15597 May 20 22:30 zabbix_agentd.conf.saved
-rw-r--r--  1 root  wheel  24172 May 20 22:14 zabbix_server.conf
drwxr-xr-x  2 root  wheel      2 May 20 21:34 zabbix_server.conf.d
-rw-r--r--  1 root  wheel   2668 May 20 21:52 zabbix_web_service.conf
✔ daffy 11:38 # ls -l /usr/local/etc/zabbix/scripts/
total 108
-rwxr-xr-x  1 root  wheel   4648 May 20 21:37 HL-3170CDW.pl
drwxr-xr-x  2 root  wheel      3 May 20 22:18 __pycache__
-rwxr-xr-x  1 root  wheel    615 Jun 11 18:38 ipmitool-wrapper
-rwxr-xr-x  1 root  wheel   3133 May 20 21:37 mini_ipmi_bsdcpu.py
-rwxr-xr-x  1 root  wheel  12026 May 20 21:37 mini_ipmi_smartctl.py
-rwxr-xr-x  1 root  wheel   7110 May 20 21:37 sender_wrapper.py
✔ daffy 11:38 # ls -l /usr/local/etc/zabbix/zabbix_agentd.conf.d/
total 0
✔ daffy 11:40 # perl -ne 's/#.*$//; print unless /^\s*$/;' /usr/local/etc/zabbix/zabbix_agentd.conf
LogFile=/var/log/zabbix/zabbix_agentd.log
LogFileSize=0
DebugLevel=3
Server=192.168.1.11,127.0.0.1
ServerActive=192.168.1.11,127.0.0.1
Hostname=daffy
Timeout = 30
Include=/usr/local/etc/zabbix/zabbix_agentd.conf.d/*.conf
LoadModulePath = /usr/local/etc/zabbix/modules
✔ daffy 11:44 # perl -ne 's/#.*$//; print unless /^\s*$/;' /usr/local/etc/zabbix/zabbix_server.conf
LogFile=/var/log/zabbix/zabbix_server.log
LogFileSize=0
DebugLevel=3
PidFile=/var/run/zabbix/zabbix_server.pid
DBName=zabbix
DBUser=zabbix
DBPassword=**************
DBPort=5432
StartIPMIPollers=3
StartTrappers=10
StartDiscoverers=6
CacheSize=32M
Timeout=30
FpingLocation=/usr/local/sbin/fping
LogSlowQueries=3000
StatsAllowedIP=127.0.0.1
✔ daffy 11:45 # /usr/local/sbin/zabbix_server --version
zabbix_server (Zabbix) 5.4.1rc1
Revision {ZABBIX_REVISION} 14 May 2021, compilation time: May 20 2021 21:32:24

Copyright (C) 2021 Zabbix SIA
License GPLv2+: GNU GPL version 2 or later <http://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it according to
the license. There is NO WARRANTY, to the extent permitted by law.

This product includes software developed by the OpenSSL Project
for use in the OpenSSL Toolkit (http://www.openssl.org/).

Compiled with OpenSSL 1.1.1h-freebsd  22 Sep 2020
Running with OpenSSL 1.1.1h-freebsd  22 Sep 2020
✔ daffy 11:55 # ifconfig igb0
igb0: flags=8943<UP,BROADCAST,RUNNING,PROMISC,SIMPLEX,MULTICAST> metric 0 mtu 1500
	options=a500b9<RXCSUM,VLAN_MTU,VLAN_HWTAGGING,JUMBO_MTU,VLAN_HWCSUM,VLAN_HWFILTER,VLAN_HWTSO,RXCSUM_IPV6>
	ether b8:ca:3a:5d:a2:54
	inet 192.168.1.11 netmask 0xffffff00 broadcast 192.168.1.255
	media: Ethernet autoselect (1000baseT <full-duplex>)
	status: active
	nd6 options=29<PERFORMNUD,IFDISABLED,AUTO_LINKLOCAL>
```
[image]

[image]

nobodysu, 2021-06-13:
Hm. If it says unsupported - it's really unsupported.
We are talking about single host right? Same releases of server and agent?
Are custom keys present in effective keys output?
`client$ zabbix_agentd -p`

packy, 2021-06-13:
Yes, same host, same releases of server and agent.

```
✔ daffy 11:56 # zabbix_agentd -p
agent.hostname                                [s|daffy]
agent.ping                                    [u|1]
agent.version                                 [s|5.4.1rc1]
system.localtime[utc]                         [u|1623601373]
web.page.get[localhost,,80]                   [t|HTTP/1.1 200 OK
Date: Sun, 13 Jun 2021 16:22:53 GMT
Server: Apache/2.4.43 (FreeBSD) PHP/7.4.6
Last-Modified: Sun, 10 May 2020 14:53:23 GMT
ETag: "2d-5a54c646299eb"
Accept-Ranges: bytes
Content-Length: 45
Content-Type: text/html

<html><body><h1>It works!</h1></body></html>]
web.page.perf[localhost,,80]                  [d|0.00058960914611816406]
web.page.regexp[localhost,,80,OK]             [s|OK]
vfs.file.size[/etc/passwd]                    [u|3175]
vfs.file.time[/etc/passwd,modify]             [u|1621997781]
vfs.file.exists[/etc/passwd]                  [u|1]
vfs.file.contents[/etc/passwd]                [t|# $FreeBSD: releng/12.2/etc/master.passwd 359447 2020-03-30 17:07:05Z brooks $
#
root:*:0:0:Charlie &:/root:/bin/csh
toor:*:0:0:Bourne-again Superuser:/root:
daemon:*:1:1:Owner of many system processes:/root:/usr/sbin/nologin
operator:*:2:5:System &:/:/usr/sbin/nologin
bin:*:3:7:Binaries Commands and Source:/:/usr/sbin/nologin
tty:*:4:65533:Tty Sandbox:/:/usr/sbin/nologin
kmem:*:5:65533:KMem Sandbox:/:/usr/sbin/nologin
games:*:7:13:Games pseudo-user:/:/usr/sbin/nologin
news:*:8:8:News Subsystem:/:/usr/sbin/nologin
man:*:9:9:Mister Man Pages:/usr/share/man:/usr/sbin/nologin
sshd:*:22:22:Secure Shell Daemon:/var/empty:/usr/sbin/nologin
smmsp:*:25:25:Sendmail Submission User:/var/spool/clientmqueue:/usr/sbin/nologin
mailnull:*:26:26:Sendmail Default User:/var/spool/mqueue:/usr/sbin/nologin
bind:*:53:53:Bind Sandbox:/:/usr/sbin/nologin
unbound:*:59:59:Unbound DNS Resolver:/var/unbound:/usr/sbin/nologin
proxy:*:62:62:Packet Filter pseudo-user:/nonexistent:/usr/sbin/nologin
_pflogd:*:64:64:pflogd privsep user:/var/empty:/usr/sbin/nologin
_dhcp:*:65:65:dhcp programs:/var/empty:/usr/sbin/nologin
uucp:*:66:66:UUCP pseudo-user:/var/spool/uucppublic:/usr/local/libexec/uucp/uucico
pop:*:68:6:Post Office Owner:/nonexistent:/usr/sbin/nologin
auditdistd:*:78:77:Auditdistd unprivileged user:/var/empty:/usr/sbin/nologin
www:*:80:80:World Wide Web Owner:/nonexistent:/usr/sbin/nologin
ntpd:*:123:123:NTP Daemon:/var/db/ntp:/usr/sbin/nologin
_ypldap:*:160:160:YP LDAP unprivileged user:/var/empty:/usr/sbin/nologin
hast:*:845:845:HAST unprivileged user:/var/empty:/usr/sbin/nologin
tests:*:977:977:Unprivileged user for tests:/nonexistent:/usr/sbin/nologin
nobody:*:65534:65534:Unprivileged user:/nonexistent:/usr/sbin/nologin
packy:*:10000:10000:Packy Anderson,PacKayve:/home/packy:/usr/local/bin/bash
messagebus:*:556:556:D-BUS Daemon User:/nonexistent:/usr/sbin/nologin
avahi:*:558:558:Avahi Daemon User:/nonexistent:/usr/sbin/nologin
_tss:*:601:601:TCG Software Stack user:/var/empty:/usr/sbin/nologin
cups:*:193:193:Cups Owner:/nonexistent:/usr/sbin/nologin
vboxusers:*:920:920:Virtualbox user:/nonexistent:/usr/sbin/nologin
vbox:*:10002:10002:User &:/home/vbox:/usr/local/bin/bash
git_daemon:*:964:964:git daemon:/nonexistent:/usr/sbin/nologin
rslsync:*:998:817:rslsync:/nonexistent:/bin/sh
ldap:*:389:389:OpenLDAP Server:/nonexistent:/usr/sbin/nologin
polkitd:*:565:565:Polkit Daemon User:/var/empty:/usr/sbin/nologin
colord:*:970:970:colord color management daemon:/nonexistent:/usr/sbin/nologin
kelonzi:*:10001:10001:Kay Koch:/home/kelonzi:/usr/local/bin/bash
zabbix:*:122:122:Zabbix NMS:/nonexistent:/bin/sh
postgres:*:770:770:PostgreSQL Daemon:/var/db/postgres:/bin/sh
cyrus:*:60:60:the cyrus mail server:/nonexistent:/usr/sbin/nologin
zabbix-agent:*:121:121:Zabbix Agent Account:/usr/lib/zabbix:/sbin/nologin
boinc:*:973:973:BOINC user:/var/db/boinc:/bin/sh
[... 35 lines cut ...]
net.if.out[lo0,bytes]                         [u|4546310157566]
net.if.total[lo0,bytes]                       [u|9092620588380]
net.if.collisions[lo0]                        [u|0]
net.if.discovery[lo0]                         [m|ZBX_NOTSUPPORTED] [Item does not allow parameters.]
vm.memory.size[free]                          [u|146899529728]
proc.num[inetd]                               [u|0]
proc.mem[inetd]                               [u|0]
system.cpu.switches                           [u|2168601212]
system.cpu.intr                               [u|2143774790]
system.cpu.util[all,user,avg1]                [m|ZBX_NOTSUPPORTED] [Collector is not started.]
system.cpu.load[all,avg1]                     [d|0.44287109375]
system.cpu.num[online]                        [u|32]
system.cpu.discovery                          [m|ZBX_NOTSUPPORTED] [Collector is not started.]
system.uname                                  [s|FreeBSD daffy 12.2-RELEASE-p6 FreeBSD 12.2-RELEASE-p6 GENERIC amd64]
system.sw.arch                                [s|amd64]
system.swap.size[all,free]                    [u|17179869184]
system.uptime                                 [u|2943469]
system.boottime                               [u|1620657904]
net.tcp.service[ssh,127.0.0.1,22]             [u|1]
net.tcp.service.perf[ssh,127.0.0.1,22]        [d|0.008365631103515625]
net.udp.service[ntp,127.0.0.1,123]            [u|1]
net.udp.service.perf[ntp,127.0.0.1,123]       [d|0.0001]
system.hostname                               [s|daffy]
✔ daffy 12:22 #
```

nobodysu, 2021-06-13:
```
✔ daffy 11:38 # ls -l /usr/local/etc/zabbix/zabbix_agentd.conf.d/
total 0
```
Well, that's it.
Maybe user parameters got deleted during update. Or different dir been used previously.

packy, 2021-06-13:
Thank you!
```
✔ daffy 12:33 # ls -ld /usr/local/etc/zabbix
lrwxr-xr-x  1 root  wheel  23 May 20 22:01 /usr/local/etc/zabbix -> /usr/local/etc/zabbix54
✔ daffy 12:33 # cp -v /usr/local/etc/zabbix50/zabbix_agentd.conf.d/* /usr/local/etc/zabbix54/zabbix_agentd.conf.d/
/usr/local/etc/zabbix50/zabbix_agentd.conf.d/speedtest-ookla.conf -> /usr/local/etc/zabbix54/zabbix_agentd.conf.d/speedtest-ookla.conf
/usr/local/etc/zabbix50/zabbix_agentd.conf.d/template_db_postgresql.conf -> /usr/local/etc/zabbix54/zabbix_agentd.conf.d/template_db_postgresql.conf
/usr/local/etc/zabbix50/zabbix_agentd.conf.d/userparameter-HL-3170CDW.conf -> /usr/local/etc/zabbix54/zabbix_agentd.conf.d/userparameter-HL-3170CDW.conf
/usr/local/etc/zabbix50/zabbix_agentd.conf.d/userparameter-ipmitool.conf -> /usr/local/etc/zabbix54/zabbix_agentd.conf.d/userparameter-ipmitool.conf
/usr/local/etc/zabbix50/zabbix_agentd.conf.d/userparameter_mini-ipmi2.conf -> /usr/local/etc/zabbix54/zabbix_agentd.conf.d/userparameter_mini-ipmi2.conf
/usr/local/etc/zabbix50/zabbix_agentd.conf.d/zfs.conf -> /usr/local/etc/zabbix54/zabbix_agentd.conf.d/zfs.conf
✔ daffy 12:34 # service zabbix_agentd restart
Stopping zabbix_agentd.
Waiting for PIDS: 58379 58381 58382 58384 58385 58386 58387.
Starting zabbix_agentd.
✔ daffy 12:35 # zabbix_get -s 192.168.1.11 -k mini.disktemp.discovery[get,"Example host"]
{
    "data": [
        {
            "{#DISK}": "pass0"
        },
        {
            "{#DISK}": "pass1"
        },
        {
            "{#DISK}": "pass2"
        },
        {
            "{#DISK}": "pass3"
        },
        {
            "{#DISK}": "pass4"
        },
        {
            "{#DISK}": "pass5"
        },
        {
            "{#DISK}": "pass6"
        },
        {
            "{#DISK}": "pass7"
        }
    ]
}
```

closed 2021-06-13 by packy