nobody43/zabbix-mini-IPMI #70: Introduce zabbix-mini-IPMI with Windows 10

Manero-Taube-Monica, 2022-01-13:
I have followed all the steps to implement the configuration but it does not work. Maybe it is because I have a Windows 10 on my client. Do you know if it could work on Windows 10? the hardwaremonitor application makes queries. Maybe it is the python script that is not working properly. I hope you can help me and guide me. Best regards.

Manero-Taube-Monica, 2022-01-14:
The output on my client W10
[image]

nobodysu, 2022-01-14:
Hmm, could you post your OHMR output also? (omitting the serials)
`> OpenHardwareMonitorReport.exe > ohmr.txt`

jpaulickcz, 2022-02-01:
Here's mine. However it seems to work just fine on Windows 10 for me. (Just looked in here out of curiosity)

[image]

[ohmr.txt](https://github.com/nobodysu/zabbix-mini-IPMI/files/7979362/ohmr.txt)

nobody43, 2023-02-23:
Tested with everything. Unable to reproduce.

closed 2023-02-23 by nobody43