nobody43/zabbix-mini-IPMI #71: No CPUs were found for temperature test (mini-IPMI)

BassT23, 2022-09-10:
**Please complete the following information:**
 - OS: Ubuntu 22.04
 - Zabbix server version: 5.2.7
 - Active or passive check: Active
 - Using zabbix proxy: No proxy
**Additional context**
Raspberry Pi 4 B 4GB

nobodysu, 2022-09-10:
Do you have an output on:
```
$ cat /sys/class/hwmon/hwmon0/temp1_input
```

Then try using:
```
UserParameter=mini.cputemp.discovery[*], echo '{"data": [{"{#CPU}": 0 }]}' && echo \"$2\" mini.cpu.temp[cpu0,MAX] $(( `cat /sys/class/hwmon/hwmon0/temp1_input` / 1000 )) | zabbix_sender -c /etc/zabbix/zabbix_agentd.conf -i - &> /dev/null
```

BTW, what's your `sensors -u` output?

BassT23, 2022-09-11:
hwmon output is:
`58913`

sensors -u:
```cpu_thermal-virtual-0
Adapter: Virtual device
temp1:
  temp1_input: 58.913

rpi_volt-isa-0000
Adapter: ISA adapter
in0:
  in0_lcrit_alarm: 0.000```

sry, I'm no expert in zabbix programming. I set  the code in /etc/zabbix/zabbix_agentd.conf.d/userparameter_mini-ipmi2.conf (third line) and restarted the agent service. Is this correct? Do anything else?

nobodysu, 2022-09-11:
If there's no errors - all should be fine. You can test it like the others:
https://github.com/nobodysu/zabbix-mini-IPMI#testing