nobody43/zabbix-mini-IPMI #72: Executing check on client

michlalp, 2023-02-06:
Hello, please what is the supposed way of executing check on client, currently items are type  "trapper" so I should use some scheduler to execute script on client every minute?

nobody43, 2023-02-07:
Hi.
You have to run the script this way:
`client$ python /etc/zabbix/scripts/mini_ipmi_lmsensors.py get "Example host"`
With task scheduler or cron.
If it involves smartctl - it must be under admin privileges because of disk access.

Or, it might be more convenient to change it from the server: `Template mini-IPMI v2 active` -> `Discovery list/CPU discovery` -> `Update interval`.

But don't go with low values - the data might clog up. You might want to lower [`TIMEOUT`](https://github.com/nobody43/zabbix-mini-IPMI/blob/master/Linux/mini_ipmi_lmsensors.py#L46) (delay).

nobody43, 2023-02-23:
Hope that's solved. Let me know if you have other questions.

closed 2023-02-23 by nobody43