nobody43/zabbix-mini-IPMI #80: How to solve "No CPUs were found for temperature test (mini-IPMI)"?

yeroslaviz, 2024-02-12:
**Describe the problem**
After following the instructions and adding all the files I'm getting the problem, that no CPUs can be found using the parameters provided. 

The discovery became supported, but I get the message:

```
No CPUs were found for temperature test (mini-IPMI)
```

When testing the installation locally, like described in the repository, this is what I get:

```
# /etc/zabbix/scripts/mini_ipmi_lmsensors.py get 10.33.0.158
{
    "data": []
}

# /etc/zabbix/scripts/mini_ipmi_lmsensors.py getverb 10.33.0.158
...
  Data sent to zabbix sender:

"10.33.0.158" mini.cpu.info[ConfigStatus] "NOGPUS, NOCPUS"
zabbix_sender [607789]: DEBUG: answer [{"response":"success","info":"processed: 0; failed: 1; total: 1; seconds spent: 0.000023"}]
Response from "10.33.0.65:10051": "processed: 0; failed: 1; total: 1; seconds spent: 0.000023"
sent: 1; skipped: 0; total: 1
```

**Provide all outputs described in [Testing](https://github.com/nobodysu/zabbix-mini-IPMI#testing) step**
Serial numbers should be replaced with X_SERIAL_X.

**Please complete the following information:**
 - OS: Ubunru 22.04
 - Zabbix server version: 6.0.8
 - Active or passive check: passive and active check are allowed with agent version 5.0.27
 - Using zabbix proxy:  using no proxy

**Additional context**

```
# cat /sys/class/hwmon/hwmon0/temp1_input
24625
root@supercub017:~# sensors -u
bnxt_en-pci-6301
Adapter: PCI adapter
temp1:
  temp1_input: 70.000

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:
  temp1_input: 24.750
Tccd1:
  temp3_input: 25.000
Tccd3:
  temp5_input: 23.000
Tccd5:
  temp7_input: 24.250
Tccd7:
  temp9_input: 22.750

bnxt_en-pci-6300
Adapter: PCI adapter
temp1:
  temp1_input: 70.000

k10temp-pci-00cb
Adapter: PCI adapter
Tctl:
  temp1_input: 27.500
Tccd1:
  temp3_input: 25.750
Tccd3:
  temp5_input: 26.250
Tccd5:
  temp7_input: 26.500
Tccd7:
  temp9_input: 27.250
```

I also tried to solution you posted in [71](https://github.com/nobody43/zabbix-mini-IPMI/issues/71), but I can't use it on my system, as it doesn't let the agent start.

any Ideas?

thanks

Assa

nobody43, 2024-02-12:
Hi.
Is it two cpus? What's their names?

yeroslaviz, 2024-02-16:
Hi again,

just wanted to mention, I have tried the template on a different server. Here it could identify the CPUs and show me the temperature.

Any ideas by now, why it doesn't wrk with the one above?

nobody43, 2024-02-16:
This is because the script does not support EPYC's `sensors` layout currently. I'm working on a fix.

yeroslaviz, 2024-02-19:
Also, another question.

can you maybe tell me, why is it, I can only measure temprature of CPUs, but not disks or motherboards? I know I don't have GPUs on the server, but what do I need to change in order for disks to be redable? 

I have a raid structure, but I'm not sure, where to change it in the script to be readable.

thanks

[image]

nobody43, 2024-02-20:
What's your `# smartctl --scan`?
How do you get SMART output from your drives with `smartctl`?

yeroslaviz, 2024-02-20:
The output 

```
# smartctl --scan
/dev/sda -d scsi # /dev/sda, SCSI device
/dev/sdb -d scsi # /dev/sdb, SCSI device
/dev/bus/0 -d megaraid,0 # /dev/bus/0 [megaraid_disk_00], SCSI device
/dev/bus/0 -d megaraid,1 # /dev/bus/0 [megaraid_disk_01], SCSI device
/dev/bus/0 -d megaraid,2 # /dev/bus/0 [megaraid_disk_02], SCSI device
/dev/bus/0 -d megaraid,3 # /dev/bus/0 [megaraid_disk_03], SCSI device
/dev/bus/0 -d megaraid,4 # /dev/bus/0 [megaraid_disk_04], SCSI device
/dev/bus/0 -d megaraid,5 # /dev/bus/0 [megaraid_disk_05], SCSI device
/dev/bus/0 -d megaraid,6 # /dev/bus/0 [megaraid_disk_06], SCSI device
/dev/bus/0 -d megaraid,7 # /dev/bus/0 [megaraid_disk_07], SCSI device
/dev/bus/0 -d megaraid,8 # /dev/bus/0 [megaraid_disk_08], SCSI device
/dev/bus/0 -d megaraid,9 # /dev/bus/0 [megaraid_disk_09], SCSI device
/dev/bus/0 -d megaraid,10 # /dev/bus/0 [megaraid_disk_10], SCSI device
/dev/bus/0 -d megaraid,11 # /dev/bus/0 [megaraid_disk_11], SCSI device
/dev/bus/0 -d megaraid,12 # /dev/bus/0 [megaraid_disk_12], SCSI device
/dev/bus/0 -d megaraid,13 # /dev/bus/0 [megaraid_disk_13], SCSI device
/dev/bus/0 -d megaraid,14 # /dev/bus/0 [megaraid_disk_14], SCSI device
/dev/bus/0 -d megaraid,22 # /dev/bus/0 [megaraid_disk_22], SCSI device
/dev/bus/0 -d megaraid,23 # /dev/bus/0 [megaraid_disk_23], SCSI device
```

To get sepecific information from the drives, I need to include the megaraid argument, so it is as such:

```
sudo smartctl -x /dev/sda -d megaraid,1
# or for all:
for i in {0..23}; do sudo smartctl -x /dev/sda -d megaraid,$i ;done # >>./OUTPUT; done
```

yeroslaviz, 2024-02-20:
BTW, would the template also work with agent2 of zabbix?

nobody43, 2024-02-26:
Does
`# smartctl -x /dev/bus/0 -d megaraid,1`
outputs anything?

>BTW, would the template also work with agent2 of zabbix?

It works if I'm not mistaken.

yeroslaviz, 2024-02-26:
> > BTW, would the template also work with agent2 of zabbix?
> 
> It works if I'm not mistaken.

Would this mean, I just need to cahnge the path to the agent in the mini_ipmi_smartctl.py script?

from:

```
agentConf_LINUX    = r'/etc/zabbix/zabbix_agentd.conf'
```
to 

```
agentConf_LINUX    = r'/etc/zabbix/zabbix_agent2.conf'
```

and restart the services?

yeroslaviz, 2024-02-27:
> Edit: You might need to also change binary path.

I'm not sure, what you mean. I can execute the command `smartctl`

```
# which smartctl
/usr/sbin/smartctl

# smartctl -h 
smartctl 7.2 2020-12-30 r5155 [x86_64-linux-5.15.0-84-generic] (local build)
Copyright (C) 2002-20, Bruce Allen, Christian Franke, www.smartmontools.org

Usage: smartctl [options] device

============================================ SHOW INFORMATION OPTIONS =====

...
```

So do I need to change the path to the `bin` file?

> Regarding SAS, have you changed [this setting](https://github.com/nobody43/zabbix-mini-IPMI/blob/32246a4ced7ff069b1f4eee749445f03f8ba8fd6/mini_ipmi_smartctl.py#L27)?

I set this parameter to `True`

It now works also with the agent2, at least for the CPU temperature. 
Do I need to change anything about the Raid in the scripts?

Any news about the EPYC's sensors?

thanks for all the help

nobody43, 2024-02-28:
> Do I need to change anything about the Raid in the scripts?

I'm puzzled why it doesn't work automatically after setting `isCheckSAS`. Try providing manual configuration here:
https://github.com/nobody43/zabbix-mini-IPMI/blob/32246a4ced7ff069b1f4eee749445f03f8ba8fd6/mini_ipmi_smartctl.py#L47
I will solve this inconsistency during future refactoring.

> Any news about the EPYC's sensors?

Always occupied, sorry! Try this fix:
https://github.com/nobody43/zabbix-mini-IPMI/blob/01c8b27e77545ce501c67fad15d2339a00782997/Linux/mini_ipmi_lmsensors.py#L41

yeroslaviz, 2024-02-28:
It might be a good idea to split this issue into two, as it has two different topics. The EPYC sensors and the raid structure

yeroslaviz, 2024-02-28:
> > Do I need to change anything about the Raid in the scripts?
> 
> I'm puzzled why it doesn't work automatically after setting `isCheckSAS`. Try providing manual configuration here:
> 
> https://github.com/nobody43/zabbix-mini-IPMI/blob/32246a4ced7ff069b1f4eee749445f03f8ba8fd6/mini_ipmi_smartctl.py#L47
> 

For the Raid manual configuration - Do you mean, I should add all the raid HDDs into the line: e.g.

```
diskDevsManual = ['/dev/sda -d sat+megaraid,0', '/dev/sda -d sat+megaraid,1', ... '/dev/sda -d sat+megaraid,4', '/dev/sda -d sat+megaraid,5', ... '/dev/sda -d sat+megaraid,23']
```

Like the output from the command `smartctl --scan` I gave you above?

> I will solve this inconsistency during future refactoring.

Not sure what it means 🤔

yeroslaviz, 2024-02-28:
> > Any news about the EPYC's sensors?
> 
> Always occupied, sorry! Try this fix:
> 
> https://github.com/nobody43/zabbix-mini-IPMI/blob/01c8b27e77545ce501c67fad15d2339a00782997/Linux/mini_ipmi_lmsensors.py#L41

This apply to a different server with an EPYC processor. 

I added the line to the list of reg expressions

```
CORES_REGEXPS = (
    ('Core(?:\s+)?(\d+):\n\s+temp\d+_input:\s+(\d+)'),
    ('Core(\d+)\s+Temp:\n\s+temp\d+_input:\s+(\d+)'),
    ('Tdie:\n\s+temp(\d+)_input:\s+(\d+)'),
    ('Tccd(\d+):\n\s+temp\d+_input:\s+(\d+)'),
    ('k\d+temp-pci-\w+\nAdapter:\s+PCI\s+adapter\ntemp(\d+):\n\s+temp\d+_input:\s+(\d+)'),
)
```
but it still doesn't show any reading

nobody43, 2024-02-29:
> For the Raid manual configuration - Do you mean, I should add all the raid HDDs into the line: e.g.

Try your output first, with less drives:
`diskDevsManual = ['/dev/bus/0 -d megaraid,0', '/dev/bus/0 -d megaraid,1']`

> Not sure what it means 🤔

Proper automatic fix won't happen soon.

> Now I can see the readings from the CPU and disks here. really, thanks a lot for all the help.

Great! Congratulations!

> Is there a way to also see the temperature of the motherboard?

Are you talking about `bnxt_en-pci-*`? Are you sure 70C is a reliable reading?

yeroslaviz, 2024-03-05:
Still not working, not sure why.

I added the changes as discussed

```

# Manually provide disk list or RAID configuration if needed.
#diskDevsManual = []
# like this:
diskDevsManual = ['/dev/bus/0 -d megaraid,0', '/dev/bus/0 -d megaraid,1', '/dev/bus/0 -d megaraid,2','/dev/bus/0 -d megaraid,3']
# more info: https://www.smartmontools.org/wiki/Supported_RAID-Controllers
```

But now, when I try to execute the testing command I get an error:

```
# /etc/zabbix/scripts/mini_ipmi_smartctl.py  get "supercube008"
Traceback (most recent call last):
  File "/etc/zabbix/scripts/mini_ipmi_smartctl.py", line 381, in <module>
    scanErrorNotype = scanErrors[0]
IndexError: list index out of range
```
`hostname` is supercube008

yeroslaviz, 2024-03-05:
I don't know about the motherboard, but won't be surprised if it is correct. They get very hot sometimes.

I can now see the Disk temperatures, though it is still the default `diskDevsManual = []` setting. I can't say, why it didn't work before.

nobody43, 2024-04-09:
https://github.com/nobody43/zabbix-mini-IPMI/pull/83
Have you been able to try this?

I've looked into `bnxt_en-pci-*`, it's a network adapter(s). It appears your `sensors` output does not have motherboard sensors. `sensors-detect` might help with that.

nobody43, 2024-06-12:
Hope that's solved.

closed 2024-06-12 by nobody43